raspiblitz/raspiblitz #1406: Chenge

rootzoll, 2020-08-02:


rootzoll, 2020-08-02:
The idea is to basicall switch more to a GitFlow process instead working all the time on a branch release.

So the idea would be:
- development is done on the `dev` branch
- new features should be done on single feature branches and merged into dev once ready.
- when a release comes closer, a new release branch gest created - release candidate sd card images are available
- once a release is ready, that release branch will be set as the "default" branch on GitHub (so its shown to a fresh user)
- hot fixes will be created as single hotfix branchs and once ready will be merged into the latest release branch and the dev branch

This way the sd card release images are still connected with their matching release branch and by just using `patch` command they sync with that branch and get the hotfixes deliviered

frennkie, 2020-08-02:
I support this..! @openoms objections?

openoms, 2020-08-02:
No objections. It all makes sense. 👍

rootzoll, 2020-08-04:
Just a thought: If we get rid if the "master" branch and instead switching which release branch will be the "default" to display ... will that make it harder for people on GitHub to follow the project (from the view of the github build in social functions)?

openoms, 2020-08-18:
Maybe keep the `master` branch to contain the equivalent from the latest SDcard release. 
Development in `dev`
Forked to versioned branches for RC images.
Releases (including minor ones) to be tagged from `master ` to keep clear history.

frennkie, 2020-09-05:
> Just a thought: If we get rid if the "master" branch and instead switching which release branch will be the "default" to display

In my opinion it should be the branch of the currently released SD card (currently `v1.6`). Makes much more sense for me that the current practice of showing `master` which might have many changes (also to readme.md).

> Maybe keep the master branch to contain the equivalent from the latest SDcard release.

We discussed this and the latest SDcard state should be in the branch for that release (e.g. `v1.6`)

@rootzoll so you now created `v1.7` branch (which I wouldn't have) but not `dev` .. how should we continue? :-D

frennkie, 2020-09-05:
Ok.. I merged `v1.7` into `master`, created `dev` and removed `v1.7`.

rootzoll, 2020-09-08:
Yeah I think everybody is confused now ... please lets make a group call and get all in sync on this.

rootzoll, 2020-09-08:
I always thought that a v1.6 branch will always work with a v1.6 sd card build and thats how the "hot update" works because the git on a v1.6 RaspiBlitz is set to the v1.6 branch. 

And master is the latest "released code" but now there are already changes for v1.7 in there .. that even will not be part of v1.6.1.

I can understand that changing to a more "common model" is easier for the rest and I am open to that ... but now in the middle of perparing v1.6.1 I dont know anymore where to merge what. @openoms @frennkie please lets make a video call soon to get on the same page here again. Thanks.

frennkie, 2020-09-08:
> Today we deleted the "master" branch on the RaspiBlitz GitHub repository .. the default branch is now always the latest release branch. We did it mostly because of practical reasons ... but also because we dont believe in masters ;)
https://twitter.com/rootzoll/status/1303404320105402376?s=20 

\o/ 👍 

I think we agreed that we need to be "disciplined" in two aspects:

* if changes (e.g. hotfixes) are made to older release branches (e.g. `v1.6`) then those change should immediately also be included in `dev` 
* after a branch for a future release (e.g. `v1.7`) is created we should agree on which changes should go into both `dev` and the new branch and which changes are for the future (and should therefore still be keeped in a separate feature branch).

rootzoll, 2020-09-10:
To make it easier to test PRs on a RaspiBlitz I crreated this PR: #1531

frennkie, 2020-09-10:
How about an "extra"/hidden menu where certain things are that "normal" people don't need... but which are useful for contributors and the repo owner..?!

rootzoll, 2020-09-14:
I think the update > patch-menu is already like a "hidden menu" normal users will not be bothered with. To hide it even more is maybe a bit too much. I would either make it easier to call it directly with a terminal command like "github" that brings you directly in the menu where you find all "contributer" options.

rootzoll, 2020-09-16:
Now using branches for developing features and testing on my raspiblitz I added the "github" command to quickly jump to the menu to canhe branches etc. Quite usefull. Will merge into v1.6.1

rootzoll, 2020-10-08:
OK .. as final part of v1.6.1 release document the new process and the github options.

commit 5e14b272 (2020-10-19): fix #1406 documentation
commit 4563ada4 (2020-10-19): fix #1406 documentation
closed 2020-10-19 by rootzoll