raspiblitz/raspiblitz #1408: ip2tor: shutdown 3.127.188.50 "aws1"

frennkie, 2020-08-02:
Currently there some active bridges on 3.127.188.50 "aws1". The plan is to phase out this host.

Would be a good case to check the handling of subscriptions when a host is deleted and all bridges go away.

grnqrtr, 2020-08-09:
My subscription/bridge wasn't working, so I checked github and found this thread. I had a "aws1" bridge so I cancelled it.

When I try to create a new subscription, none of the three options work for me. This is what I get:

[image]

Here is the output one time from details:
```
###### ERROR DETAIL FOR DEBUG #######

Error Short:
failed HTTP code
Shop:
https://ip2tor.fulmo.org
Bridge:
{'id': 'de5f9e86-dd35-4a75-88e8-652f91cd2330', 'site': 'ip2tor.fulmo.org', 'created_at': '2020-07-03T10:07:55Z', 'modified_at': '2020-08-09T11:56:22Z', 'ip': '91.109.21.148', 'is_enabled': True, 'is_alive': True, 'name': 'eu-fra1', 'is_testnet': False, 'offers_tor_bridges': True, 'tor_bridge_duration': 86400, 'tor_bridge_price_initial': 40000, 'tor_bridge_price_extension': 24000, 'offers_rssh_tunnels': False, 'rssh_tunnel_price': 1000, 'terms_of_service': '-', 'terms_of_service_url': '-', 'ci_date': '2020-08-09T11:56:22Z', 'ci_message': '', 'ci_status': 0, 'owner': 4, 'tor_bridge_duration_hours': 24, 'tor_bridge_price_initial_sats': 40, 'tor_bridge_price_extension_sats': 24}
Error Detail:
{'status_code': 504, 'timestamp': '2020-08-09T11:57:41Z'}

Press Enter to continue ...
```

frennkie, 2020-08-09:
There is maintenance ongoing on the ip2tor.fulmo.org host (since yesterday). The machine is running but very slow. This also revealed some bugs. Please try again later today (may be 1-2 hours).

frennkie, 2020-08-13:
@grnqrtr thanks for moving off `aws1`. Sorry for the inconvenience last weekend. But since late Sunday evening everything should be running smoothly. Can you confirm?

grnqrtr, 2020-08-14:
No problem! Yes, I moved to one of the other options several days ago and things have been running smoothly.

closed 2020-08-14 by frennkie