raspiblitz/raspiblitz #1411: Specter 0.6.0

rootzoll, 2020-08-05:
see https://github.com/rootzoll/raspiblitz/issues/1171#issuecomment-669027062 from #1171

ben-kaufman, 2020-08-14:
Hi, in case you haven't started with this yet, there's version 0.6.1 which contains a few bug fixes and would be nice to have instead.

https://github.com/cryptoadvance/specter-desktop/releases/tag/v0.6.1

rootzoll, 2020-08-14:
@ben-kaufman thx, havent yet 0.6.1 sounds good :)

rootzoll, 2020-08-24:
Also merged PR #1481 to add missing library

ChuckNorrison, 2020-08-29:
Here we go :)
https://github.com/cryptoadvance/specter-desktop/releases/tag/v0.7.2

rootzoll, 2020-09-02:
Update PR merged - moving to final testing.

openoms, 2020-09-19:
Specter 0.8.0 is out:
https://github.com/cryptoadvance/specter-desktop/releases/tag/v0.8.0

Update successfully with the UPDATE menu (available in the latest script for RaspiBlitz v1.6.1)

I suggest we should ship with the latest version by default.

openoms, 2020-09-22:
​Update form Specter v0.5.5 in RaspiBlitz v1.6:
```
#uninstall old version
/config.scripts/bonus.cryptoadvance-specter.sh off
# install the missing dependency
sudo apt install -y libffi-dev
# change script to use the latest version
sed -i "s#cryptoadvance.specter==0.5.5#cryptoadvance.specter==0.7.2#g"  /home/admin/config.scripts/bonus.cryptoadvance-specter.sh

rootzoll, 2020-09-28:
Update Bug Reported: https://github.com/rootzoll/raspiblitz/pull/1566#issuecomment-699681734

rootzoll, 2020-10-05:
@openoms merged #1619 ... did you found any hint of the bug mentioned above while testing.

openoms, 2020-10-05:
Nope, but I am way past of updating from 0.5.5. 
There were no errors installing v0.8.0 or 0.8.1. Update works too.

openoms, 2020-10-07:
We could probably add this entry to the bitcoin.conf:
[image]

Should bring some performance gains and have enough space on the 1TB drives.
Anyone concerns?

```
checking how long it takes, just started now on my RPi4 4GB + SSD:
```
$ sudo cat  /mnt/hdd/bitcoin/debug.log | grep "block filter"
2020-10-07T19:00:12Z * Using 111.8 MiB for basic block filter index database
2020-10-07T19:00:54Z basic block filter index thread start
2020-10-07T19:00:54Z Syncing basic block filter index with block chain from height 0
2020-10-07T19:01:25Z Syncing basic block filter index with block chain from height 6824
...
2020-10-07T19:03:32Z Syncing basic block filter index with block chain from height 15584
```

openoms, 2020-10-08:
Was about 10 hours in the background rescanning a couple wallets simultaneously. 
```
sudo cat  /mnt/hdd/bitcoin/debug.log | grep "block filter"
2020-10-07T19:00:12Z * Using 111.8 MiB for basic block filter index database
2020-10-07T19:00:54Z basic block filter index thread start
2020-10-07T19:00:54Z Syncing basic block filter index with block chain from height 0
...
2020-10-08T05:15:31Z basic block filter index is enabled at height 651757
2020-10-08T05:15:31Z basic block filter index thread exit
```
Once activated the warning from Specter is gone, so does not need ,much explanation I think.

rootzoll, 2020-10-08:
All funtions of Specter run OK while the additional index is building ... or do we need a info in the status menu, that the "index is still building"?

openoms, 2020-10-08:
I did not find any problems while building the index as it works without as well only slightly slower. The warning disappears probably because it is looking in the bitcoin.conf.
Will ask quickly in the Specter TG to be sure.

rootzoll, 2020-10-11:
OK merged the Specter-BlockfilterIndex PR ... installs fine so far. I am see this testing OK. @openoms if you dont get any contradicting info from the specter TG then I will close this issue for the v1.6.1 release.

openoms, 2020-10-11:
they said there should be no issue while indexing. Specter queries the bitcoind RPC for everything, that should be able to deal with it.

rootzoll, 2020-10-11:
OK perfect .. closing this issue for v.16.1 release.

closed 2020-10-12 by rootzoll