raspiblitz/raspiblitz #1416: Create an additional HS for rpcport 18332

Fonta1n3, 2020-08-06:
Generally most people want to use Fully Noded on testnet first, this means they need to manually add a HS to the raspiblitz torrc file for port 18332 which can present some challenges for users who are not familiar with setting up directories, permissions and owners via cli.

I have found through my personal experience its really nice to create a dedicated HS for each rpcport that way a user can not only use multiple chains simultaneously but also connect to them remotely via dedicated hidden services simultaneously. The UX is really great being able to switch between networks seamlessly.

For example FN has some more advanced wallet features around multisig that people really ought to test out first before committing real funds.

Raspiblitz users have not been able to connect FN when the node is on testnet bc the HS is of course for `rpcport` 8332.

Sorry for not submitting a PR to add this and being lazy to raise an issue! Just something to consider that I think users would appreciate.

anonymoustrident, 2020-08-06:
Thank You @Fonta1n3 for assisting with the troubleshooting and putting this PR here.

rootzoll, 2020-08-09:
- [ ] check HiddenService for testnet RPC port

Fonta1n3, 2020-08-10:
Ooops, I thought your comment was meant to say this was already a feature :)

keblek, 2021-03-09:
@Fonta1n3 @rootzoll since this update includes Core 0.21 then adding signet would be useful as well. Especially since clightning supports signet.

rootzoll, 2021-03-25:
ok I added on the same hidden service address for BITCOIN RPC the ports for testnet 18332 & signet 38332
this will be part of the v1.7RC1 release - ready for final testing

rootzoll, 2021-04-13:
@openoms the additional ports seem to be removed - see old changes:
https://github.com/rootzoll/raspiblitz/commit/b5222647aa8c36f302bfe5cd2ac05685f9dc5a5d

Is that due to the Tor refactoring?

openoms, 2021-04-13:
yes, I think serving additional hidden services by default (especially to mainnet bitcoin rpc) when it is bot used is not desired.
8332 now has a menu option in CONNECT to be opened on demand.

That should be used to create the service for testnet as well.

Signet is not used anywhere so far, but agree that it should be supported in a future version.

closed 2021-04-18 by rootzoll