raspiblitz/raspiblitz #1420: RTL not loading after update 1.5.3 to 1.6.0

Smiggel, 2020-08-06:
I noticed that when RTL runs on the old version of Raspiblitz, it does not work after upgrading to 1.6.0. I did see in the logs that it was installed and enabled, but the webinterface is not accesible.

I removed RTL using the menu, and re-installed it using the menu and then it works.

rootzoll, 2020-08-06:
As candle reported there was an outage at NPM .. future versions will make better "error detection" on the installs to notify the user on it: https://github.com/rootzoll/raspiblitz/pull/1419

rootzoll, 2020-08-13:
Merged the NPM install check by candle. This should be a good enough quick fix until we can make installs more ressilient in general.

- [ ] Check Install of RTL again

rootzoll, 2020-10-08:
OK tested RTL install again - is working fine. Closing this issue for v1.6.1 release.

closed 2020-10-08 by rootzoll