raspiblitz/raspiblitz #1425: Joinmarket local order book w/ Tor

woeisme, 2020-08-08:
When installing local orderbook service on joinmarket and tor is enabled the following dialogue is blank:
[image]

woeisme, 2020-08-08:
Additional information from ob-watcher
[image]

rootzoll, 2020-08-09:
@openoms ?

openoms, 2020-08-09:
try to run:
`pip3 install matplotlib`
at the prompt: ` (jmvenv) joinmarket`
the install is quite lengthy and might fail.
Will look into it.

openoms, 2020-08-09:
The output should be improved, but the ob-watcher works for me.
@woeisme exit the log screen with CTRL+C and select the `OFFERS` option again. 
It should show you the Hidden Service address for you own Offer Book.

Here is my one tested:
`gopnmsknawlntb4qpyav3q5ejvvk6p74a7y5xotmph4v64wl3wicscad.onion`

openoms, 2020-08-11:
Also might be related: https://github.com/JoinMarket-Org/joinmarket-clientserver/issues/657#issuecomment-669344536
Can try 
`sudo apt-get install libopenjp2-7 libtiff5`
if not solved when selecting the `OFFERS` option in the menu again.

woeisme, 2020-08-11:
Hey @openoms 

I followed your instructions, this time it displays an onion address correctly, but visiting that address with Tor Browser shows 'Unable to connect'. 

Next page after that is the complaint page about missing matplotlib again. 

Trying:

 `sudo apt-get install libopenjp2-7 libtiff5`

resulted in: 

```
sudo apt-get install libopenjp2-7 libtiff5
Reading package lists... Done
Building dependency tree
Reading state information... Done  
libopenjp2-7 is already the newest version (2.3.0-2+deb10u1).  
libtiff5 is already the newest version (4.1.0+git191117-2~deb10u1).  
The following package was automatically installed and is no longer required:  
  libbluetooth3  
Use 'sudo apt autoremove' to remove it.  
```

Still not showing offers however.

openoms, 2020-08-11:
can you try the manual instructions?

https://github.com/openoms/bitcoin-tutorials/tree/master/joinmarket#run-the-offer-book-locally

Not sure if I can reproduce the problem.

woeisme, 2020-08-15:
using:
`sudo apt-get install python-matplotlib`

seems to have installed it, however running `python ob-watcher.py` either from cli or from within the menu still fails at detecting it is installed.

openoms, 2020-08-15:
@woeisme try:
`sudo apt-get install python3-matplotlib`

openoms, 2020-08-15:
@woeisme reworked the `OFFERS` in JoininBox to fix the above problems.
You can use the `UPDATE` option to get to the latest master.

closed 2020-08-15 by openoms

woeisme, 2020-08-18:
I had attempted the pip install of matplotlib earlier with no success.

Went into main Raspiblitz service menu, removed joinmarket then reinstalled. 
Switched to joinmarket user ssh session, First selected Update, then Offers -> Start (start watching locally) which installed matplotlib successfully. 
Order book now loads locally.

openoms, 2020-08-18:
Great, thank you for reporting!