raspiblitz/raspiblitz #1427: Error while formatting HDD

jayrmotta, 2020-08-09:
## Problem

While configuring a new node from scratch (SD and HDD clean) an error occurred during the HDD formatting phase

## Steps to reproduce

1. Burn raspiblitz 1.6.0 on your SD
2. Format your HDD (did it on my macOS)
3. Start the raspiblitz configuration process
4. Select bitcoin + lightning
5. Setup passwords A and B
6. Wait to see the problem

## Hardware

Raspberry Pi 3b+
1tb Samsung USB 3 HDD

## Logs

```
# --> Formatting HDD/SSD ...
# DATA DRIVE - FORMATTING to new EXT4 layout (old)
# Unmounting all partitions of this device
# Cleaning /etc/fstab from `-sda1
# Wiping all partitions (sfdisk/wipefs)
Error: Input/output error during write on /dev/sda
# Creating the one big partition
# waiting until the partion gets available
# waiting until the partion gets available
# waiting until the partion gets available
# waiting until the partion gets available
# waiting until the partion gets available
# waiting until the partion gets available
# waiting until the partion gets available
# waiting until the partion gets available
# waiting until the partion gets available
# waiting until the partion gets available
# waiting until the partion gets available
/dev/fd/63: line 2: The: command not found
/dev/fd/63: line 3: syntax error near unexpected token `('
/dev/fd/63: line 3: `Calling ioctl() to re-read partition table.'
# OK

# *** 40addHDD.sh ***
# FAIL given device not found
FAIL blitz.datadrive.sh tempmount --> device not found
Please report issue to the raspiblitz github.
```

I hope it helps, let me know if I can provide more info

rootzoll, 2020-08-09:
@jayrmotta thanks for the report. Adding this to the v.16.1 update to take alook:

The `${uuid}` seems to have the wrong value ... seeing this from this log line:
```
cleaning /etc/fstab from `-sda1
```
- [ ] check why the uuid gets discovered wrongly

apruden2008, 2020-09-05:
I've got the same issue trying to build new from a Raspberry Pi 3b

# --> Formatting HDD/SSD ...
# DATA DRIVE - FORMATTING to new EXT4 layout (old)
# Unmounting all partitions of this device
# Cleaning /etc/fstab from f6a2611b-3755-4e8c-92de-18ded213ff56
# Wiping all partitions (sfdisk/wipefs)
Re-reading the partition table failed.: Device or resource busy
wipefs: error: /dev/sda: probing initialization failed: Device or resource busy
# WARNING: partitions are still not clean - try Quick & Dirty
1+0 records in
1+0 records out
512 bytes copied, 0.000449525 s, 1.1 MB/s
/dev/fd/63: line 2: The: command not found
/dev/fd/63: line 3: syntax error near unexpected token `('
/dev/fd/63: line 3: `Calling ioctl() to re-read partition table.'
# OK

# *** 40addHDD.sh ***
# FAIL given device not found
FAIL blitz.datadrive.sh tempmount --> device not found
Please report issue to the raspiblitz github.

apruden2008, 2020-09-05:
I've tried a few times and also seen this variant: 

```
mke2fs 1.44.5 (15-Dec-2018)
mkfs.ext4: Input/output error while writing out and closing file system
# waiting until formatted drives gets available
```

johngribbin, 2020-09-07:
Hey folks, 

I am having what looks like a similar issue. I have a brand new raspi 4 and a Samsumg 1TB SSD. Here is the error I am getting in the console: 

```
# --> Formatting HDD/SSD ...
# DATA DRIVE - FORMATTING to new EXT4 layout (old)
# Unmounting all partitions of this device
# Wiping all partitions (sfdisk/wipefs)
# Creating the one big partition
# waiting until the partion gets available
# Formatting
mke2fs 1.44.5 (15-Dec-2018)
# waiting until formatted drives gets available
# waiting until formatted drives gets available
# waiting until formatted drives gets available
# waiting until formatted drives gets available
# waiting until formatted drives gets available
# waiting until formatted drives gets available
# waiting until formatted drives gets available
# waiting until formatted drives gets available
# waiting until formatted drives gets available
# waiting until formatted drives gets available
# waiting until formatted drives gets available
# FAIL blitz.datadrive.sh format --> formatting ext4 failed
# Please report issue to the raspiblitz github.
```

Anyone got any ideas? Thanks in advance.

rootzoll, 2020-09-21:
This this also because of update on `blitz.datadrive.sh` script - see:  #1297

rootzoll, 2020-09-23:
I am not completly sure why these error happen - the test setup iof @jayrmotta is the one I do regularly to test releases. What Format are you using on MacOS to clean the HDD?

I tried to make the partition wiping more robust, added more debug info and will now test this again for the final v1.6.1 release.

rootzoll, 2020-10-06:
TODO: Because of #1628 when everything else works - run one test with BTRFS.

rootzoll, 2020-10-15:
Testing:
- [ ] HDD with Apple Format "GUID Partionstabelle"
- [ ] HDD with Apple Format "Master Boot Record"
- [ ] HDD with Apple Format "Apple-Partionstabelle"
- [ ] HDD with all deleted
- [ ] HDD try with BTRFS

rootzoll, 2020-10-16:
All tests went well .. just the BTRFS test showed a bug to be fixed for v1.7 #1676
Closing this issue for v1.6.1 release.

diskbuster, 2020-12-07:
Hi guys,
i am not sure if I am at the right place with my issue, but the logfile on 1.6.2 told me so:

# ERROR: formatting ext4 failed
/dev/fd/63: line 2: The: command not found
/dev/fd/63: line 3: syntax error near unexpected token `('
/dev/fd/63: line 3: `Calling ioctl() to re-read partition table.'
# OK

# *** 40addHDD.sh ***

# --> Linking drives/directories
# hddCandidate='sda'
# hddPartitionCandidate='sda1'
FAIL blitz.datadrive.sh link --> no data drive mounted
Please report issue to the raspiblitz github.

Is there a workaround to format my ssd for use it with the one big partition for the blockchain?

openoms, 2020-12-07:
yes, just format the disk on your desktop to any filesystem and make sure there is only one partition on it.

diskbuster, 2020-12-07:
thank you. tried efs4 format on raspi before, that seems to be the problem.
fat32 format on Mac solves the problem and setup can proper format and use my ssd now.

ezeql, 2021-03-23:
Hitting the same issue as the @diskbuster. Tried formatting on another system using EXT4 and FAT32 with the same outcome

STEMr42, 2021-07-31:
Hello, I found a source of the formatting errors in the original post.  The 'label' is not generating for lsblk on the partition when running the blitz.datadrive.sh script.

line 645
`       loopdone=$(lsblk -o NAME,LABEL | grep -c BLITZDATA)`
line 685
`       loopdone=$(lsblk -o NAME,LABEL | grep -c BLITZSTORAGE)`
and line 724
`       loopdone=$(lsblk -o NAME,LABEL | grep -c BLITZTEMP)`

The loop times out because the label doesn't show with lsblk.  For some reason, calling parted will refresh it and the label will show correctly with lsblk.  I added a line above each of these to call parted again and the errors cleared.

`sudo parted -l`

rootzoll, 2021-08-17:
@STEMr42 thanks added the `sudo parted -l` lines. Will be part of v1.7.1 RC1 release.

nu5h4ckle, 2021-08-22:
@STEMr42 i added `sudo parted -l` below each line and it also solved the error.  kudos!

rootzoll, 2021-10-09:
Formatting HDD tested fine during RC1 & RC2 .. closing for final release.

closed 2021-10-09 by rootzoll