raspiblitz/raspiblitz #1428: bonus.electrsexplorer.sh restarts BTC-RPC-Explorer every 10 minutes if electrs is also enabled (v1.6.0)

PatrickScheich, 2020-08-09:
looking into **_background.sh** around line 430

```
  ###############################
  # Set the address API use for BTC-RPC-Explorer depending on Electrs status
  ###############################

  # check every 10 minutes
  electrsExplorer=$((($counter % 600)+1))
  if [ ${electrsExplorer} -eq 1 ]; then
    if [ "${BTCRPCexplorer}" = "on" ] & [ "${ElectRS}" = "on" ]; then
      /home/admin/config.scripts/bonus.electrsexplorer.sh
    fi
  fi
```
according the variables **BTCRPCexplorer** and **ElectRS** sourced from **raspiblitz.conf**, the script **bonus.electrsexplorer.sh** will be called approx. every 10minutes

A) there is a small issue in the _here_document_ 
```
      # create ExecStart=/home/admin/btc-rpc-explorer.run.sh     
      cat > /home/admin/btc-rpc-explorer.run.sh <<EOF
#!/bin/bash
echo "Waiting Electrs on port 50001..."
while [ $(sudo -u electrs lsof -i | grep -c 50001) -eq 0 ]; do
. . .
EOF
```

the **$(sudo..** parts need an extra quoting

```
      # create ExecStart=/home/admin/btc-rpc-explorer.run.sh     
      cat > /home/admin/btc-rpc-explorer.run.sh <<EOF
#!/bin/bash
echo "Waiting Electrs on port 50001..."
while [ \$(sudo -u electrs lsof -i | grep -c 50001) -eq 0 ]; do
. . .
EOF
```
should do the trick 

B) the logic of the script leads to a restart of the BTC-RPC-Explorer service in any case
I guess that is not desired. I have done a quick fix, which works for me.
File is attached, feel free to use it if you consider it as helpful
[bonus.electrsexplorer.sh.txt](https://github.com/rootzoll/raspiblitz/files/5048167/bonus.electrsexplorer.sh.txt)

rootzoll, 2020-08-09:
good catch ... will take a look at this for v1.6.1 patch/service update

rootzoll, 2020-09-16:
@PatrickScheich testing your script ... looks good - much thx.

rootzoll, 2020-09-18:
Merged into v1.6 branch - so can be tested with MENU > UPDATE > PATCH > PATCHMENU > PATCH
Once switch EletcRS and Explorer on/off ... because I changed the place where the start script gets stored.

rootzoll, 2020-10-14:
OK in my tests it runs now without restart. Will be part of v1.6.1 release. Closing issue.

closed 2020-10-14 by rootzoll