raspiblitz/raspiblitz #1429: Lightning Progress: prepare sync

JurGi7, 2020-08-10:
I updated to 1.6.0. However, the flash has remained for 3 days with a blue screen and the status Node is Syncing:
Blockchain Progress: 100%
Lightning Progress: **prepare sync**
Please wait - this can take some time ...
I tried to restart, what else can I do?

bitsam, 2020-08-10:
Exit to command prompt and type debug. Then post the output in reply.

JurGi7, 2020-08-10:
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.6.0
chainnetwork: bitcoin / main
 09:20:15 up  3:27,  2 users,  load average: 1.67, 1.59, 1.56

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-08-10 05:55:48 BST; 3h 24min ago
  Process: 3022 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 3027 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 3066 (bitcoind)
    Tasks: 18 (limit: 2200)
   Memory: 641.2M
   CGroup: /system.slice/bitcoind.service
           └─3066 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/…

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Mon 2020-08-10 08:19:52 BST, end at Mon 2020-08-10 09:20:15 BST. --
-- No entries --

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-08-10T05:59:04Z UpdateTip: new best=000000000000000000007cdd53042ac9ea3240b0255a3fcc07dde7fdfc5af0b7 height=643034 version=0x20000000 log2_work=92.18899 tx=556960373 date='2020-08-10T05:58:56Z' progress=1.000000 cache=8.3MiB(73674txo) warning='74 of last 100 blocks have unexpected version'
2020-08-10T05:59:42Z UpdateTip: new best=000000000000000000039c97f7e47d29a7989f58a772785862af1f50a9c0211f height=643035 version=0x20000000 log2_work=92.189008 tx=556961634 date='2020-08-10T05:59:04Z' progress=1.000000 cache=8.4MiB(74807txo) warning='73 of last 100 blocks have unexpected version'
2020-08-10T06:31:04Z Potential stale tip detected, will try using extra outbound peer (last tip update: 1882 seconds ago)
2020-08-10T06:31:10Z New outbound peer connected: version: 70015, blocks=643035, peer=573 (full-relay)
2020-08-10T06:34:56Z UpdateTip: new best=0000000000000000000f9b41b3edf2645f00fcbaea8d60b87d2cd2d5d828107c height=643036 version=0x20000000 log2_work=92.189027 tx=556964295 date='2020-08-10T06:34:21Z' progress=1.000000 cache=10.8MiB(97129txo) warning='73 of last 100 blocks have unexpected version'
2020-08-10T06:45:33Z Pre-allocating up to position 0x6000000 in blk02189.dat
2020-08-10T06:45:33Z UpdateTip: new best=0000000000000000000ea5a9d06c374ffb8e4e92130c8c9b6509b21d1cc9a5cf height=643037 version=0x20400000 log2_work=92.189045 tx=556965857 date='2020-08-10T06:45:29Z' progress=1.000000 cache=12.7MiB(110427txo) warning='73 of last 100 blocks have unexpected version'
2020-08-10T06:50:21Z UpdateTip: new best=00000000000000000007b0958b14cdbafe1a3c74ea95be81670e0d2635284777 height=643038 version=0x2fffe000 log2_work=92.189064 tx=556968075 date='2020-08-10T06:50:12Z' progress=1.000000 cache=13.4MiB(117108txo) warning='74 of last 100 blocks have unexpected version'
2020-08-10T06:50:25Z ERROR: AcceptBlockHeader: Consensus::ContextualCheckBlockHeader: 000000000000000000639be19a0123a1c99d9fef89f0b8ac055a77f4ef86ae3b, bad-diffbits, incorrect proof of work
2020-08-10T06:58:05Z UpdateTip: new best=00000000000000000006b118d6dafcaf65eb20ec7152be997d0b84ea888d3ea6 height=643039 version=0x20400000 log2_work=92.189082 tx=556970046 date='2020-08-10T06:57:56Z' progress=1.000000 cache=14.1MiB(123735txo) warning='74 of last 100 blocks have unexpected version'
2020-08-10T07:05:10Z Pre-allocating up to position 0xc00000 in rev02189.dat
2020-08-10T07:05:10Z UpdateTip: new best=0000000000000000000fc2455dd1b7a16eb951db7888834e566e33718f2a1cda height=643040 version=0x37ffe000 log2_work=92.1891 tx=556972361 date='2020-08-10T07:04:58Z' progress=1.000000 cache=14.8MiB(130016txo) warning='74 of last 100 blocks have unexpected version'
2020-08-10T07:10:49Z UpdateTip: new best=0000000000000000000e9ffc2bc0230ddd0d0e76753af0de53c22af96f54299c height=643041 version=0x20000000 log2_work=92.189119 tx=556973717 date='2020-08-10T07:10:45Z' progress=1.000000 cache=15.2MiB(134130txo) warning='73 of last 100 blocks have unexpected version'
2020-08-10T07:23:38Z UpdateTip: new best=0000000000000000000fed1f0d7727879989992e2c810745d6bc6d8c50f81d1c height=643042 version=0x20400000 log2_work=92.189137 tx=556976096 date='2020-08-10T07:23:35Z' progress=1.000000 cache=16.2MiB(143581txo) warning='74 of last 100 blocks have unexpected version'
2020-08-10T07:31:00Z ping timeout: 1200.042171s
2020-08-10T07:31:06Z UpdateTip: new best=00000000000000000005dbfe9a7127c92e6583d05aabb030dfdcdf446a83f20e height=643043 version=0x20c00000 log2_work=92.189156 tx=556977727 date='2020-08-10T07:30:46Z' progress=1.000000 cache=16.9MiB(149693txo) warning='74 of last 100 blocks have unexpected version'
2020-08-10T07:44:14Z UpdateTip: new best=000000000000000000094ebbef725a509bcd165db1a88661a87ac7007db6a126 height=643044 version=0x20400000 log2_work=92.189174 tx=556979867 date='2020-08-10T07:43:48Z' progress=1.000000 cache=18.0MiB(160442txo) warning='75 of last 100 blocks have unexpected version'
2020-08-10T07:56:12Z Pre-allocating up to position 0xd00000 in rev02189.dat
[... cut ...]
initialSync=0
syncProgress=100.00 
startcountLightning=1
lndActive=1
lndRPCReady=0
walletOpened=1
walletReady=0
walletLocked=0
# LND RPC is still warming up - no scan progress: prepare scan
scanTimestamp=-2
syncedToChain=0
lndRPCReady=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=2

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

bitsam, 2020-08-10:
Your node is running a little hot and has reported HW errors, which look to be temp warnings. The rpi seems a little sensitive to power and heat and sometimes leads to corrupt files on the disk. I would look into some cooling for the pi. I have mine sitting on top of an old 5V pc fan (run of separate power). I am running v1.6 without any current issues. I did have the error with the index and decided to delete the bitcoin data and re-sync the entire bitcoin chain data and index (copied from my PC).

Another thing I have started doing before a Raspiblitz upgrade is to disable RTL, as this makes the upgrade process longer. 

@rootzoll , is there a script @JurGi7 could run to remove RTL, to see if it will take some load off the pi?

Also is it possible fo change the order of services being enabled when performing a Raspiblitz version upgrade that RTL could be upgraded up last, so the Node is back up and fully operational, then RTL is upgraded/enabled in the background?

JurGi7, 2020-08-11:
Hello, thanks for the hint. Now my raspiblitz is working again. Have a fan built in. The temperature is now 44 degrees. Unfortunately I have to do without the display now. Thx bitsam ;-)

[image]

rootzoll, 2020-08-13:
> @rootzoll , is there a script @JurGi7 could run to remove RTL, to see if it will take some load off the pi?

You can switch off RTL in the SERVICES submenu - see if this changes something.

> Also is it possible fo change the order of services being enabled when performing a Raspiblitz version upgrade that RTL could be upgraded up last, so the Node is back up and fully operational, then RTL is upgraded/enabled in the background?

We have much longer taking installs the RTL .. I will refactor the install of apps and features with the next version ... but every install would still need to finish before it boots into normal mode.

I will close this issue ... but re-open if you see any further bug to fix in here.

closed 2020-08-13 by rootzoll