raspiblitz/raspiblitz #1430: Incorrect password instruction for Specter

shawnyeager, 2020-08-10:
Per instructions, below, user is to use Password B, whereas Password A is what successfully logged me in.

[image]

rootzoll, 2020-08-10:
@shawnyeager Thanks for the report ... wil recheck for update release v1.6.1

rootzoll, 2020-09-22:
OK I just rechecked with the lastest code for v1.6.1 release ... it is password B for specter login.

Last check before closing the issue: Test on a fresh setuped RaspiBlitz with test v1.6.1 code

rootzoll, 2020-10-08:
OK found some smaller bugs to fix on fnal testing: #1642 .. but Specter looks good now and used all the correct passwordB on tests. Closing this issue for v1.6.1 release.

closed 2020-10-08 by rootzoll