raspiblitz/raspiblitz #1432: At first setup start a wifi ap and show access info on display.

Emzy, 2020-08-10:
I have an idea for improvement.
For the first start after flashing the sd card start a wifi access point so you can set your wifi password.
This is oftern used on wifi micro controllers or other embedded devices.
 
- First check if we get an IP address via DHCP in the ethernet interface.
- If not start up host ap with ssid="RaspiBlitz-[mac address]" password="[random]"
- show wifi ssid and password on LCD.

If this is a good idea I will look into it.
Asking for feedback.

rootzoll, 2020-08-13:
This idea is matching what you can see in this design for the startup screen i am aming fot with a upcoming release as the "WIFI" button- see: https://github.com/rootzoll/raspiblitz/issues/993

But the "WIFI" was more of an additional idea we pushed a but back to consider once the TOR and SSH options work fine.
So the idea is good just not on high priority. @Emzy but feel free to alraedy make some reasearch and post it here how to activate a Wifi-Access point by command line and how to make nginx serve a special webpage just to that network device.

nyxnor, 2021-05-17:
@Emzy Any news?