raspiblitz/raspiblitz #1434: Incorrect password instruction for BTC-RPC-Explorer [release 1.6.1]

shawnyeager, 2020-08-11:
User is instructed to use Password B, but Password A is required.

[image]

rootzoll, 2020-08-13:
@shawnyeager I checked ... it should be password B ... I found a bug that if you change password B afterwards some other passwords on apps dont get changed. So could it be that you had in the beginning the same password A as B and then changed B?

shawnyeager, 2020-08-13:
@rootzoll Ah, interesting. I set all passwords (using 1Password's generator) at time time of install, and have not changed. For whatever reason, I've only ever been able to login to web services with Password A.

rootzoll, 2020-08-13:
Can you take a look into your bitcoin conf and just let me know that there is really your password B (and not your Password A that maybe got entered during setup becaue of a copy paste hickup). Todo so do ... go to Terminal and run:
`sudo cat /mnt/hdd/bitcoin/bitcoin.conf` and verify whats set for `rpcpassword`.

rootzoll, 2020-08-13:
Seems related: https://github.com/rootzoll/raspiblitz/issues/1445

rootzoll, 2020-10-08:
I just tested again ... its Password B as instructed. Closing issue for v1.6.1 release.

closed 2020-10-08 by rootzoll