raspiblitz/raspiblitz #1435: V1.6 no QR code for mobile connect

byob75, 2020-08-11:
Hello,

I am not getting a QR code when I try ZAP or Zeus for android.
Tried display and console.

I'm running behind TOR.

Looks like it is not connecting for a download.
Do I need to open a port on my firewall?

Thanks

-----------------------------------------------------------------------------------------

               RaspiBlitz v1.6.0  [_removed_]
               Bitcoin Fullnode + Lightning Network + Tor
        ,/     -------------------------------------------
      ,'/      CPU load 0.45, 0.60, 0.81, temp 66°C 151°F
    ,' /       Free Mem 3259M / 3906M  HDDuse 346G (40%)
  ,'  /_____,  SSH admin@192.168.1.129 d13.3MiB u9.0MiB
 .'____    ,'  Web admin --> http://192.168.1.129:3000
      /  ,'    bitcoin v0.20.0 mainnet Sync OK 100.00%
     / ,'      Public [removed] 10 peers
    /,'
   /'          LND 0.10.4-beta wallet [_removed_] sat
               2/2 Channels [_removed_] sat 5 peers
               Fee Report: 0-0-0 sat (D-W-M)

--------------------------------------------------------------------------------------------------------------

creating install info ...
# yes-button -> TOR
# Installing lndconnect..
Check Framework: Go

Installed go version go1.13.3 linux/arm

package golang.org/x/sys/unix: unrecognized import path "golang.org/x/sys/unix" (https fetch: Get https://golang.org/x/sys/unix?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:48749->192.168.1.1:53: i/o timeout)
package golang.org/x/crypto/ripemd160: unrecognized import path "golang.org/x/crypto/ripemd160" (https fetch: Get https://golang.org/x/crypto/ripemd160?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:59955->192.168.1.1:53: i/o timeout)
package golang.org/x/crypto/ed25519: unrecognized import path "golang.org/x/crypto/ed25519" (https fetch: Get https://golang.org/x/crypto/ed25519?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:33532->192.168.1.1:53: i/o timeout)
package golang.org/x/net/ipv4: unrecognized import path "golang.org/x/net/ipv4" (https fetch: Get https://golang.org/x/net/ipv4?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:38378->192.168.1.1:53: i/o timeout)
package golang.org/x/net/ipv6: unrecognized import path "golang.org/x/net/ipv6" (https fetch: Get https://golang.org/x/net/ipv6?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:44032->192.168.1.1:53: i/o timeout)
package golang.org/x/net/proxy: unrecognized import path "golang.org/x/net/proxy" (https fetch: Get https://golang.org/x/net/proxy?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:45599->192.168.1.1:53: i/o timeout)
HEAD is now at 82d7103 Merge pull request #15 from Kixunil/patch-1
go get ./...
package golang.org/x/sys/unix: unrecognized import path "golang.org/x/sys/unix" (https fetch: Get https://golang.org/x/sys/unix?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:50862->192.168.1.1:53: i/o timeout)
package golang.org/x/crypto/ripemd160: unrecognized import path "golang.org/x/crypto/ripemd160" (https fetch: Get https://golang.org/x/crypto/ripemd160?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:51815->192.168.1.1:53: i/o timeout)
package golang.org/x/crypto/ed25519: unrecognized import path "golang.org/x/crypto/ed25519" (https fetch: Get https://golang.org/x/crypto/ed25519?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:47648->192.168.1.1:53: i/o timeout)
package golang.org/x/net/ipv4: unrecognized import path "golang.org/x/net/ipv4" (https fetch: Get https://golang.org/x/net/ipv4?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:42882->192.168.1.1:53: i/o timeout)
package golang.org/x/net/ipv6: unrecognized import path "golang.org/x/net/ipv6" (https fetch: Get https://golang.org/x/net/ipv6?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:52428->192.168.1.1:53: i/o timeout)
package golang.org/x/net/proxy: unrecognized import path "golang.org/x/net/proxy" (https fetch: Get https://golang.org/x/net/proxy?go-get=1: dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:45333->192.168.1.1:53: i/o timeout)
make: *** [Makefile:4: dep] Error 1
# using TOR --> host [_removed_].onion port 10009
/home/admin/config.scripts/bonus.lndconnect.sh: line 267: lndconnect: command not found
error='file does not exist'
/home/admin/config.scripts/bonus.lndconnect.sh: line 302: lndconnect: command not found
(To shrink QR code: OSX->CMD- / LINUX-> CTRL-) Press ENTER when finished.

byob75, 2020-08-11:
Is there a manual/cmd line way to link a wallet as a work around?

rootzoll, 2020-08-13:
> `/home/admin/config.scripts/bonus.lndconnect.sh: line 302: lndconnect: command not found`

Hmm so something went wrong with your LNDconnect install.
Please run `MAINMENU > UPADTE > PATCH > Patch Menu > PATCH`and try again

Let me know the results.

byob75, 2020-08-14:
Seems like it is still failing the install.

creating install info ...
# yes-button -> TOR
# Installing lndconnect..
Check Framework: Go

Installed go version go1.13.3 linux/arm

package golang.org/x/sys/unix: unrecognized import path "golang.org/x/sys/unix"                                                                                                              (https fetch: Get https://golang.org/x/sys/unix?go-get=1: dial tcp: lookup golan                                                                                                             g.org on 192.168.1.1:53: read udp 192.168.1.129:45848->192.168.1.1:53: i/o timeo                                                                                                             ut)
package golang.org/x/crypto/ripemd160: unrecognized import path "golang.org/x/cr                                                                                                             ypto/ripemd160" (https fetch: Get https://golang.org/x/crypto/ripemd160?go-get=1                                                                                                             : dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:57881->1                                                                                                             92.168.1.1:53: i/o timeout)
package golang.org/x/crypto/ed25519: unrecognized import path "golang.org/x/cryp                                                                                                             to/ed25519" (https fetch: Get https://golang.org/x/crypto/ed25519?go-get=1: dial                                                                                                              tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:38666->192.168                                                                                                             .1.1:53: i/o timeout)
package golang.org/x/net/ipv4: unrecognized import path "golang.org/x/net/ipv4"                                                                                                              (https fetch: Get https://golang.org/x/net/ipv4?go-get=1: dial tcp: lookup golan                                                                                                             g.org on 192.168.1.1:53: read udp 192.168.1.129:34672->192.168.1.1:53: i/o timeo                                                                                                             ut)
package golang.org/x/net/ipv6: unrecognized import path "golang.org/x/net/ipv6"                                                                                                              (https fetch: Get https://golang.org/x/net/ipv6?go-get=1: dial tcp: lookup golan                                                                                                             g.org on 192.168.1.1:53: read udp 192.168.1.129:45641->192.168.1.1:53: i/o timeo                                                                                                             ut)
package golang.org/x/net/proxy: unrecognized import path "golang.org/x/net/proxy                                                                                                             " (https fetch: Get https://golang.org/x/net/proxy?go-get=1: dial tcp: lookup go                                                                                                             lang.org on 192.168.1.1:53: read udp 192.168.1.129:33675->192.168.1.1:53: i/o ti                                                                                                             meout)
HEAD is now at 82d7103 Merge pull request #15 from Kixunil/patch-1
go get ./...
package golang.org/x/sys/unix: unrecognized import path "golang.org/x/sys/unix"                                                                                                              (https fetch: Get https://golang.org/x/sys/unix?go-get=1: dial tcp: lookup golan                                                                                                             g.org on 192.168.1.1:53: read udp 192.168.1.129:34680->192.168.1.1:53: i/o timeo                                                                                                             ut)
package golang.org/x/crypto/ripemd160: unrecognized import path "golang.org/x/cr                                                                                                             ypto/ripemd160" (https fetch: Get https://golang.org/x/crypto/ripemd160?go-get=1                                                                                                             : dial tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:42484->1                                                                                                             92.168.1.1:53: i/o timeout)
package golang.org/x/crypto/ed25519: unrecognized import path "golang.org/x/cryp                                                                                                             to/ed25519" (https fetch: Get https://golang.org/x/crypto/ed25519?go-get=1: dial                                                                                                              tcp: lookup golang.org on 192.168.1.1:53: read udp 192.168.1.129:33266->192.168                                                                                                             .1.1:53: i/o timeout)
[... cut ...]                                                                                                            meout)
make: *** [Makefile:4: dep] Error 1

!!!!!!!!!!!!!!!!!!!!!!!!!!
FAIL: Was not able to install/build lndconnect
Retry later or report to developers with logs above.
/home/admin/config.scripts/bonus.lndconnect.sh: line 55: lndconnect: command not                                                                                                              found
!!!!!!!!!!!!!!!!!!!!!!!!!!

byob75, 2020-08-14:
Seems like it is failing on 
  go get -d github.com/rootzoll/lndconnect
in
 raspiblitz/home.admin/config.scripts/bonus.lndconnect.sh 

I was wondering if my firewall was blocking this connection
so I performed this command "go get -d github.com/rootzoll/lndconnect"
on my desktop and it worked correctly.

rootzoll, 2020-08-14:
Maybe your GO install is broken.

Can you run on the Terminal: `/home/admin/config.scripts/bonus.go.sh off` --> it should say `OK Go removed.`
Then type `menu` and try the the mobile pairing again?

byob75, 2020-08-14:
The go uninstall was successful.
Rerunning the mobile pairing then reinstalled go.
However I am still getting the timeout errors.

Searching through the repo, I think this is the first time my node would have tried to use the "go get...."
Is it used anywhere else during the install and setup?

creating install info ...
# yes-button -> TOR
# Installing lndconnect..
Check Framework: Go
*** Installing Go v1.13.3 for armv6l ***
--2020-08-14 15:13:51--  https://dl.google.com/go/go1.13.3.linux-armv6l.tar.gz
Resolving dl.google.com (dl.google.com)... 172.217.4.78, 2607:f8b0:4009:805::200                                                                             e
Connecting to dl.google.com (dl.google.com)|172.217.4.78|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 98269277 (94M) [application/octet-stream]
Saving to: ‘go1.13.3.linux-armv6l.tar.gz’

go1.13.3.linux-armv 100%[===================>]  93.72M  5.63MB/s    in 16s

2020-08-14 15:14:12 (5.77 MB/s) - ‘go1.13.3.linux-armv6l.tar.gz’ saved [98269277                                                                             /98269277]

mkdir: cannot create directory ‘/usr/local/gocode’: File exists

Installed go version go1.13.3 linux/arm

package golang.org/x/sys/unix: unrecognized import path "golang.org/x/sys/unix"                                                                              (https fetch: Get https://golang.org/x/sys/unix?go-get=1: dial tcp: lookup golan                                                                             g.org on 192.168.1.1:53: read udp 192.168.1.129:32832->192.168.1.1:53: i/o timeo                                                                             ut)

byob75, 2020-08-14:
The lookup timeout makes me wonder if this is a DNS issue.

byob75, 2020-08-14:
Nevermind, I can ping g.org from the blitz

byob75, 2020-08-14:
Does TOR mess with GO DNS lookup?

rootzoll, 2020-08-14:
It shouldnt .. it works on my machine running behind TOR.

rootzoll, 2020-08-14:
> mkdir: cannot create directory ‘/usr/local/gocode’: File exists

I added `sudo rm -rf /usr/local/gocode` when go will be turned off - just in case.
Run MAINMENU > UPADTE > PATCH > Patch Menu > PATCH to update to the latest change if you want.

byob75, 2020-08-14:
I was missing the Network Setup -> Local DNS under Router IP in DD-WRT on my router.

I had DNS setup under DHCP, but that must not work here.

Thanks for your help and Thanks for all your hard work on this project!

closed 2020-08-14 by byob75

zanderisrael, 2020-08-28:
Still having this issue, maybe its iptables blocking "go get" requests?