raspiblitz/raspiblitz #1436: White Screen after every Boot, maybe because of newest Kernel?

phenomeus, 2020-08-11:
Hi everybody, 

I have a white screen that is not going away, i think its dead (but still powered on?) or maybe that came after kernel updates etc. can‘t say for sure, i noticed it too late. 

Any idea how to debug and get it to work again? I already tried to power down the pi and removed the power cord. 

Thanks in advance

phenomeus, 2020-08-11:
It looks like a kernel and driver issue 

https://github.com/goodtft/LCD-show/issues/223

phenomeus, 2020-08-11:
There is a fork that is updated to work with newest kernel, trying right now myself, will report back. 

```
sudo rm -rf LCD-show
git clone https://github.com/MrYacha/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show
```

phenomeus, 2020-08-11:
Rotation activation by menu brings back white screen. Will debug further

phenomeus, 2020-08-12:
I created two pull requests. My screen is working again and is rotated correctly. 

(Sorry for 2 patches, I’m still learning how to use github ;) )

phenomeus, 2020-08-13:
I have to rotate the touchscreen also, 180 degree

phenomeus, 2020-08-13:
Touch screen rotation also fixed and tested. Logic had to be turned around. PR is online

rootzoll, 2020-08-17:
Thanks .. hope to find time to review your changes in the next days.

frennkie, 2020-08-17:
I have this issue too.

frennkie, 2020-08-17:
@phenomeus wouldn't the three PRs better be put into one..? Would be great if you could look into that.

Would also be nice to have a more descriptive branch name... Then it would be easier for me to quickly check out yand test your proposed changes.

phenomeus, 2020-08-17:
Totally agree but I’m a GitHub newb :) seriously no idea how to merge my 3 prs and work with branches. 

Open to any guidance.

frennkie, 2020-08-17:
But you do use the command line, right?!

I general I recommend you go have a look some tutorials.. but here are the essential steps:

```
git remote  add phenomeus https://github.com/phenomeus/raspiblitz.git
git remote  add upstream https://github.com/rootzoll/raspiblitz.git
git fetch --all --prune
checkout -b fix-white-screen upstream/v1.6
git merge phenomeus/patch-2
git merge phenomeus/patch-3
git merge phenomeus/patch-4
git push -u phenomeus fix-white-screen
```

frennkie, 2020-08-17:
Are actually all changes needed? Or is this sufficient:

```
sudo sed -i "s/^dtoverlay=.*/dtoverlay=waveshare35a:rotate=90/g" /boot/config.txt
rm -rf /home/admin/LCD-show
git clone https://github.com/MrYacha/LCD-show.git /home/admin/LCD-show
sudo /home/admin/LCD-show/LCD35-show  # this will reboot
```

phenomeus, 2020-08-18:
Thanks for the input. 

First you have to use a fork for lcdshow 

Than modify to waveshare35a

Than fix rotation logic

ChuckNorrison, 2020-08-24:
testing this right now

phenomeus, 2020-08-25:
> > Thanks for the input.
> 
> Will you consolidate the PRs? Or should we do it?

Please do it, pretty busy right now

rootzoll, 2020-09-02:
OK i implemented it now that way that on boot it will check if the kernel was updated and may correct the LCD driver then.

Also the lcd scripts now should check what kernal is running and use the correct rotation setting.

I was not able to test the kernel upgrade because `sudo apt-get upgrade` resulted in:
```
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/w/webkit2gtk/libjavascriptcoregtk-4.0-18_2.28.3-2~deb10u1+rpi1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
```

So final testing is still TODO ... with upgrading kernel to see if it automatically adapts and to see if screen rotation is then working correctly.

frennkie, 2020-09-02:
404 errors on `apt-get upgrade`...? Either the source has a - yet unnoticed - (temporary or permanent) problem or you forgot to run `apt-get update`.

rootzoll, 2020-09-02:
@frennkie ah ok I tried `apt-get update` again and now the upgrade is working. Not sure why it was not working the first time. thanks.

rootzoll, 2020-09-02:
OK this looks good now. When I upgrade the kernel on a fresh v1.6.0 and then `restart` it picks up on boot that the kernel has changed and updates the drivers.

Last thing to test before release: If then a v1.6.1 (that will still have the old kernel because base sd card image will be changed on service release) will run thru good on recovering with touch screen.

rootzoll, 2020-10-12:
Recoverings with v1.6.1 and Touchscreen went well. Closing issue for v1.6.1 release.

closed 2020-10-12 by rootzoll

uninflatable, 2020-10-21:
Hi, I updated today my rasiplitz from 1.6.0 to 1.6.1 using the image: https://raspiblitz.org/raspiblitz-v1.6.1-2020-10-20.img.gz

after installing 1.6.1 I get a white screen and with SSH I can't connect any longer. Please advise

ChuckNorrison, 2020-10-21:
@uninflatable you need to recreate your sd card and boot from it regulary to get into the recovery process.

uninflatable, 2020-10-21:
@ChuckNorrison excellent - this worked fine! Thank you for your support!

ChuckNorrison, 2021-01-16:
today i saw the white screen again after some weeks untouched. my blitz worked yesterday. I was running 1.6.2 on this. Now i will reflash on 1.6.3

phenomeus, 2021-01-24:
Little headsup, I'm using this fork https://github.com/sun-23/LCD-show for kernel 5.10 

just in case you want to update here :)

Mogultalk, 2021-07-18:
> @uninflatable you need to recreate your sd card and boot from it regulary to get into the recovery process.

How??