raspiblitz/raspiblitz #1438: alternative method to check if electrs service is properly initialised and online in bonus.electrs.sh

PatrickScheich, 2020-08-11:
the following issue may be more a cosmetic kind

there is a bit of code in **bonus.electrs.sh** that checks if the electrs service is answering on the configured port (default 50001)

```
    electrumResponding=$(echo exit | telnet 127.0.0.1 50001 2>/dev/null | grep -c "Connected to")
```
this is executed every 10seconds by **00infoLCD.sh** and as "exit\r\n" is not a proper lnd JSONRPC, it spams the lnd journal

the above line may be changed to 

```
electrumResponding=$(echo '{"jsonrpc":"2.0","method":"server.ping","params":[],"id":"electrs-check"}' | netcat -w 2 127.0.0.1 50001 | grep -c "result")
```

which is a std JSONRPC call

rootzoll, 2020-08-13:
Thx. I changes the line in the elctrum bonus script. Running tests.

rootzoll, 2020-10-14:
OK tests look good. Closing for v1.6.1 release.

closed 2020-10-14 by rootzoll