raspiblitz/raspiblitz #1440: Failed to install Let's Encrypt

JurGi7, 2020-08-12:
I get the following error message when I try to install Let's Encrypt:

**Exception on Subscription

Unknown Error happened - please report to developers:
('failed HTTP request',
'https://www.duckdns.org/update?domains=XXXXX&token=XXXXXXX-XXXXX-XXXXX&ip=publicIP: XXX.XXX.XX.XXX',
ConnectionError(MaxRetryError("SOCKSHTTPSConnectionPool(host='www.duckdns.org', port=443):
Max retries exceeded with url:
/update?domain=XXXXXX&token=XXXXXXX-XXXXX-XXXXX&ip=publicIP (Caused by
New ConnectionError('<urllib3.contrib.socks.SOCKSHTTPSConnection object at 0x75cd6490>:
Failed to establish a new connection: [Errno 111] Connection refused'))")))**

Then nothing can be found under "LIST My Subscriptions".

Any solution?

JurGi7, 2020-08-12:
maybe because no server is running on 443 yet?

frennkie, 2020-08-12:
Are you running your Raspiblitz with or without TOR?

JurGi7, 2020-08-12:
without TOR

frennkie, 2020-08-12:
That's the reason this is failing. It tries to use the TOR proxy port on 127.0.0.1:9050. 

Workaround is to make it connect without TOR manually by removing the curl rc file. 

What would be our solution @rootzoll..? Detect inactive TOR and act accordingly?

rootzoll, 2020-08-12:
The `blitz.subscriptions.letsecrypt.py` script should normally just add the TOR proxy if the config is TRUE on TOR. See:

[image]

@frennkie how can we test that the config is giving the correct value?

frennkie, 2020-08-12:
Should work with `if cfg.run_behind_tor.value:`

rootzoll, 2020-08-13:
Added fix .. should be part of v1.6.1 service release.

closed 2020-10-08 by rootzoll