raspiblitz/raspiblitz #1441: Bitcoind service not running 1.5.1

ag2121-ag, 2020-08-12:
Hi,

Recently just built my node and all seemed to go fine. I sync and had it up and running. I wanted to use the chain explorer so I permissioned it but now I am getting the error "Bitcoind service is not running". I know this takes time but it has been this way for over 24 hours. I have read the other issue tickets and ran 

sudo tail -f /mnt/hdd/bitcoin/debug.log

I get the following:

```
ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=0, nPos=296920)
2020-08-12T13:57:55Z *** ThreadSync: Failed to read block 00000000202f36135a13e90c07fe3816d4ae948ff1e3fa941d3dee7bbf421604 from disk

 LoadBlockIndexDB: last block file = 2191
2020-08-12T13:58:40Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=69, size=89470443, heights=643123...643251, time=2020-08-10...2020-08-11)

```
I am unsure how to proceed.

rootzoll, 2020-08-12:
Did you used the torrent download of the blockchain in 1.5? There was a problem that the index building for BlockExplorer with that torrent blockchain - one reason why we dropped torrent blockchain download.

ag2121-ag, 2020-08-12:
yes I did. Should I just rsync? At this point it would probably be worth just getting 1.6?

rootzoll, 2020-08-12:
Yeah ... best is to call `repair` on the commandline (or choose in main menu) and then "RESET-CHAIN". Dont use the Torrent option. Once everything runs consider a update to v1.6.

ag2121-ag, 2020-08-12:
Thanks. I completely missed the torrent being removed from the release notes.

closed 2020-08-12 by rootzoll