raspiblitz/raspiblitz #1444: LND and bitcoind not serving over ipv6 since v. 1.6

janis-mueller, 2020-08-12:
Since the update to version 1.6, lnd and bitcoind do not seem to run over ipv6 at all, making it impossible to reach any of the services from the outside if not using, just as well as opening channels to other ipv6 nodes. Used to work (and run over tcp6) with raspiblitz v.1.5 and bitcoin core 19.0

output of netstat for my raspi:
sudo netstat -tulpen | grep lnd
sudo netstat -tulpen | grep bitcoind

**tcp**        0      0 0.0.0.0:10009           0.0.0.0:*               LISTEN      1002       54296      5335/**lnd**            
**tcp**        0      0 127.0.0.1:9735          0.0.0.0:*               LISTEN      1002       52520      5335/**lnd**            
**tcp**        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      1002       54297      5335/**lnd**

**tcp**        0      0 127.0.0.1:28332         0.0.0.0:*               LISTEN      1002       7908644    31296/**bitcoind**      
**tcp**        0      0 127.0.0.1:8332          0.0.0.0:*               LISTEN      1002       7908642    31296/**bitcoind**      
**tcp**        0      0 127.0.0.1:8333          0.0.0.0:*               LISTEN      1002       7908974    31296/**bitcoind**      
**tcp**        0      0 127.0.0.1:28333         0.0.0.0:*               LISTEN      1002       7908646    31296/**bitcoind**

as shown only serving via ipv4, please make both services run over ipv6 again 
thanks

PatrickScheich, 2020-08-13:
please take a look at #869

in comment https://github.com/rootzoll/raspiblitz/issues/869#issuecomment-671966121 you find some sources which will hard-switch the raspiblitz to IPv6 only

```
root@DesertRetreatLN:~# sudo netstat -tulpen | grep lnd
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      1002       81039      10396/lnd
tcp        0      0 0.0.0.0:10009           0.0.0.0:*               LISTEN      1002       81037      10396/lnd
tcp6       0      0 :::9735                 :::*                    LISTEN      1002       80575      10396/lnd
tcp6       0      0 ::1:8080                :::*                    LISTEN      1002       81040      10396/lnd
tcp6       0      0 ::1:10009               :::*                    LISTEN      1002       81038      10396/lnd
root@DesertRetreatLN:~# sudo netstat -tulpen | grep bitcoind
tcp        0      0 127.0.0.1:28332         0.0.0.0:*               LISTEN      1002       37165      3552/bitcoind
tcp        0      0 127.0.0.1:8332          0.0.0.0:*               LISTEN      1002       37163      3552/bitcoind
tcp        0      0 0.0.0.0:8333            0.0.0.0:*               LISTEN      1002       51026      3552/bitcoind
tcp        0      0 127.0.0.1:28333         0.0.0.0:*               LISTEN      1002       37167      3552/bitcoind
tcp6       0      0 :::8333                 :::*                    LISTEN      1002       51025      3552/bitcoind
```

janis-mueller, 2020-08-13:
I took the script you edited and replaced them with the original existing ones but this didn't bring any change, all the services are still running over IPv4 after a reboot, do I miss something?

PatrickScheich, 2020-08-13:
@janis-mueller please ssh into the raspibltz and do a

```
sudo journalctl -n 10000 -u background --no-pager | grep "_background.sh" > /home/admin/bg_000.log

....look into bg_000.log and remove all private informations, then do a...

cat  /home/admin/bg_000.log  |  nc termbin.com 9999
```

paste the created link here, as well do a `sudo systemctl status lnd.service` to check how lnd is started

...mine looks like...
```
admin@DesertRetreatLN:~ $ sudo systemctl status lnd.service
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-08-13 23:12:56 CEST; 1min 57s ago
  Process: 25028 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 25030 (lnd)
    Tasks: 14 (limit: 4915)
   Memory: 570.4M
   CGroup: /system.slice/lnd.service
           └─25030 /usr/local/bin/lnd --externalip=[2a02:6d40:3043:7801:1d83:4b30:d589:7534]:9735
```

janis-mueller, 2020-08-14:
vendor preset: enabled)
   Active: active (running) since Thu 2020-08-13 21:22:19 BST; 17h ago
 Main PID: 6246 (lnd)
    Tasks: 15 (limit: 4915)
   Memory: 291.1M
   CGroup: /system.slice/lnd.service
           └─6246 /usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --listen=127.0.0.1:9735`

janis-mueller, 2020-08-14:
thanks for the further help, here you go for the log: https://termbin.com/dhcf

and the status of lnd: 
```
sudo systemctl status lnd.service
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-08-13 21:22:19 BST; 17h ago
 Main PID: 6246 (lnd)
    Tasks: 15 (limit: 4915)
   Memory: 291.1M
   CGroup: /system.slice/lnd.service
           └─6246 /usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --listen=127.0.0.1:9735

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
```

accidentally comment closed this, did not mean to.

PatrickScheich, 2020-08-15:
@janis-mueller It seems that you have enabled TOR therefore the IPv6 is disabled (afaik the IPv6 may leak information which contradicts the usage of TOR) 

If you like you could disable the TOR option and retry with IPv6

janis-mueller, 2020-08-15:
> @janis-mueller It seems that you have enabled TOR therefore the IPv6 is disabled (afaik the IPv6 may leak information which contradicts the usage of TOR)
> 
> If you like you could disable the TOR option and retry with IPv6

Uninstalled the TOR service and can confirm it is working now using your scripts, lnd and bitcoind are both listening over my ipv6 address now, thank you a lot!

PatrickScheich, 2020-08-15:
good to hear, so you can close this issue

closed 2020-08-15 by janis-mueller