raspiblitz/raspiblitz #1446: [Feature request] Fee confirmation when using CASHOUT

openoms, 2020-08-13:
A user reported that paid 126 sat/byte, $10 overall when cashin out through the menu even when the mempool empty temporarily.

As higher fees are common detailing the deafult fee used would be needed with manual fee setting if it can it kept simple enough.

rootzoll, 2020-08-13:
Yes we could start to fix things here from the RaspiBlitz side ... but the basic problem should be adrressed in LND. We just call:

`lncli --chain=${network} --network=${chain}net sendcoins --sweepall --addr=${address} --conf_target=6`

And if the mempool is empty and still the fees should not be that high. But what we can do is set `--conf_target=36` to relax on the LND fee calculation. Cashout should normally not be a time critical task. And if users want it have it quicker next time free to call lncli or use a more better tool like RTL or Thunderhub. For the basic SSH menu, we should keep it simple and let LND optimize on their side.

rootzoll, 2020-08-13:
- [ ] check fees on chashout with new setting

frennkie, 2020-08-13:
Hm... I have wasted precious SAT recently when using the SSH menu... :-/

On thing to keep in mind that those transactions - if I recall correctly - support RBF. So if we go with 36 as a target and a user is 'in a hurry' they could bump the fee (with RBF or use CPFP).

rootzoll, 2020-08-13:
I already changed to conf-target 36 .. that should be a good enough quick fix to ease the pain. @frennkie if you like you can add some information at the end of the script how to bump the fee if user is in a hurry - that would be a nice to have.

frennkie, 2020-08-13:
Haven't used it yet on mainnet.. but there is:

```
lncli -n testnet wallet bumpclosefee 113eac09324d678b7ce247df978a536e14034b275bc7897a087e1b58bce35fb8:1
```

rootzoll, 2020-08-13:
But thats for channel closing ... not cashing out the onchain balance, right?

frennkie, 2020-08-13:
There is also a wallet subcommand for cashing out (sweep).

rootzoll, 2020-10-08:
So `--conf_target=36` should be good for a quickfix here. Will be part of v1.6.1 release .. closing issue.

closed 2020-10-08 by rootzoll