raspiblitz/raspiblitz #1447: Failed to install LOOP

marssystems, 2020-08-13:
RaspiBlitz v1.5.1

Tried to install LOOP from the RaspiBlitz Menu System - getting this error -

Created symlink /etc/systemd/system/multi-user.target.wants/loopd.service → /etc/systemd/system/loopd.service.
OK - the Lightning Loop service is now enabled
./bonus.loop.sh: line 85: loop: command not found
 Failed to install Lightning Loop

Tried running the bonus.loop.sh script manually (as directed) with same result.

rootzoll, 2020-08-13:
There were some reports on Problems with Loop on v1.5.1 .. that should be fixed and was restested with the lastest v1.6 release. Please try that one and re-open issue if you still have problems.

closed 2020-08-13 by rootzoll

marssystems, 2020-08-13:
So are you saying that the latest version (0.8.0) of LOOP installs and runs without any errors on RaspiBlitz v1.6 ? Has this been tested?
I had no problems installing v0.6.5  and v0.7.0 of LOOP on RaspiBlitz v1.5.1.
This 0.8.0 version is the first one to fail - maybe its a LOOP problem - worth investigating further.

rootzoll, 2020-08-13:
On the v1.6 release we still use Loop 0.6.5 - that worked well with the rest of the setup of the 1.6.

marssystems, 2020-08-13:
Well yes - LOOP v0.6.5 worked fine for me too but all these third party apps are being updated with new functionality all the time, not to mention all the bug fixes, so RaspiBlitz should be taking advantage of this.
There should be a menu option to upgrade all apps to the latest versions as and when they become available.

So how do I go back to v.0.6.5 of LOOP? The git pull downloads the latest version available. Is there a parameter I can add to get a specific version?

rootzoll, 2020-08-13:
> So how do I go back to v.0.6.5 of LOOP?

Thats why maybe just putting there a button that lets user update users all apps to the latest versions not a good idea ;)

With everything changing so fast testing what works together is part of our RaspiBlitz releases - at least we try our best. Everybody is free to try updates by themselves on their own risk - we even depend on reporting on that experience ... but as you can see on your own question, if we offer something by default we are also expected to offer down-grades. But that adds a lot of additional complexity .. there could be data migrations between vesion that cannot be reversed, etc. 

You can try to go with a fresh sd card, that will rebuild your system will all the old app versions.

marssystems, 2020-08-13:
No - a fresh SD card is not an option as then I will have 3 apps to upgrade to the latest versions - Bitcoin, LND and RTL.

There must be a simpler way to go back to v0.6.5 (or v0.7.0) - I will ask LOOP support for help and advice.

rootzoll, 2020-08-13:
Where do you end up with in the RaspiBlitz SERVICES menu turning Loop OFF and then ON again?

marssystems, 2020-08-13:
As my first message says - getting this error -

Created symlink /etc/systemd/system/multi-user.target.wants/loopd.service → /etc/systemd/system/loopd.service.
OK - the Lightning Loop service is now enabled
./bonus.loop.sh: line 85: loop: command not found
Failed to install Lightning Loop

The loopd service isn't running and calling the 'loop' command fails.

rootzoll, 2020-08-13:
Ah right you were on that old v1.5.1 and updated LND, Bitcoind and RTL even over the versions that v1.6 provides. So from the support side I can just say you would need to wait until v1.7 (what will be a while) are experiment on your own (always make a LND backup before tyring stuff out).

marssystems, 2020-08-13:
Yes I am and I don't have any problems running the latest releases of Bitcoin, LND and RTL.

I don't understand why now, the bonus.loop.sh script fails on line 85 and fails to install/start the loopd service and why the loop command does not exist no matter what version of LOOP I try to install.
The script worked fine before and there have not been any changes to it at all.

rootzoll, 2020-08-13:
Try to replace it with the bonus.loop.sh from the master branch.

marssystems, 2020-08-13:
Yes - just done it now - installing v0.6.5. Will let you know..........

marssystems, 2020-08-13:
OK - now its failing on line 83 -

admin@raspberrypi:~/config.scripts $ sudo ./bonus.loop.sh on
making sure the loop service is not running
*** INSTALL LIGHTNING LOOP ***
Check Framework: Go

Installed go version go1.13.3 linux/arm

Cloning into 'loop'...
remote: Enumerating objects: 59, done.
remote: Counting objects: 100% (59/59), done.
remote: Compressing objects: 100% (44/44), done.
remote: Total 2578 (delta 21), reused 36 (delta 15), pack-reused 2519
Receiving objects: 100% (2578/2578), 1.28 MiB | 2.26 MiB/s, done.
Resolving deltas: 100% (1669/1669), done.
HEAD is now at 9c503fc Merge pull request #245 from joostjager/bump-v0.6.5-beta

[Unit]
Description=Loopd Service
After=lnd.service
[Service]
WorkingDirectory=/home/bitcoin/loop
ExecStart=/usr/local/gocode/bin/loopd --network=mainnet
User=bitcoin
Group=bitcoin
Type=simple
KillMode=process
TimeoutSec=60
Restart=always
RestartSec=60
[Install]
WantedBy=multi-user.target

Created symlink /etc/systemd/system/multi-user.target.wants/loopd.service → /etc/systemd/system/loopd.service.
OK - the Lightning Loop service is now enabled
./bonus.loop.sh: line 83: loop: command not found
 Failed to install Lightning Loop

rootzoll, 2020-08-13:
Your go install seems not to run ... there should be a lot of `go: downloading ...` outputs

marssystems, 2020-08-13:
So what is the cause of that?

rootzoll, 2020-08-13:
What happens if you go to terminal and run

```
cd /home/bitcoin/loop/cmd
go install ./...
```

marssystems, 2020-08-13:
Well the 'go install ./...' command has been running for 10 mins now with no output to the terminal.

marssystems, 2020-08-13:
Got something now -

admin@raspberrypi:/home/bitcoin/loop/cmd $ go install ./...
# github.com/lightninglabs/loop/cmd/loop
panic: runtime error: index out of range [-1445]

rootzoll, 2020-08-13:
maybe something with your go install went wrong ... try

```
/home/admin/config.scripts/bonus.go.sh off
/home/admin/config.scripts/bonus.go.sh on
```

marssystems, 2020-08-13:
OK - done that - this is what I got -

admin@raspberrypi:/home/bitcoin/loop/cmd $ /home/admin/config.scripts/bonus.go.sh off
Check Framework: Go

Installed go version go1.13.3 linux/arm

admin@raspberrypi:/home/bitcoin/loop/cmd $ /home/admin/config.scripts/bonus.go.sh on
Check Framework: Go

Installed go version go1.13.3 linux/arm

rootzoll, 2020-08-13:
hmmm do this again:

```
sudo rm -rf /usr/local/go
/home/bitcoin/loop/cmd $ /home/admin/config.scripts/bonus.go.sh on
```

And then try the bonus loop script again.

marssystems, 2020-08-13:
OK - installing GO now - watch this space.

marssystems, 2020-08-13:
Ok - LOOP now installed with no errors.

I've just rebooted the RPi4 to get a clean start and I will check if LOOP is available in RTL again when back up and running.

rootzoll, 2020-08-13:
OK that maybe the right moment to remind that there is a way to donate to the project on my tallycoin page ;)
https://tallyco.in/s/r5lx23/

marssystems, 2020-08-13:
Sure! - no worries! - As soon as my node spends some time online making money, donations to the project will start flowing.

marssystems, 2020-08-13:
Yep! - LOOP v0.8.0 now working fine.

Who would have thought the problem was caused by a corrupt GO installation - you learn something new every day!

Thanks for your help.