raspiblitz/raspiblitz #1449: LetsEnrypt LNbits not loading anymore after restart

rootzoll, 2020-08-13:
I was setting up a LNbits over IP2TOR and LetsEncrypt .. worked so far.

Then I did a reboot of RaspiBlitz and now the browser was not able to reach the service anymore over the duckdns domain ... even tho I LNbits is running on local address and TOR (http & https).

Check again if reproducable.

rootzoll, 2020-08-13:
OK not sure why .. but trying agian after some minutes its now answering on the duckdns domain. But it was not answering in the beginning.

rootzoll, 2020-08-14:
- [ ] check with ubuntu@172.81.177.248 if there is any chaching on the Tor directory service

frennkie, 2020-08-14:
Just tested this my self. "Worked for me". But I don't doubt that there may be an issue. Any idea how to narrow this down?

frennkie, 2020-08-16:
OK, just rebooted a Raspiblitz and my IP2Tor Bridge for Btcpay was longer working. If something like this happens **always** first check whether the tor address is reachable with the Tor browser. In my case it wasn't. I stopped and started tor and then it worked again. Maybe something with the rendezvous points in Tor..?! @openoms any other ideas?

rootzoll, 2020-08-18:
I also observed some "outages" using the 172.81.177.248 bridge. Was not able to "connect" by using the domain - while LNbits was reachable trhu TOR. Took some minutes and then it was back again.

rootzoll, 2020-08-24:
Outages seem rrealted to this Tor error message:
```
Aug 24 05:32:46.000 [warn] Giving up on launching a rendezvous circuit to $03910F285A33F365838EC66EF2C2EF754D046760~ at 148.251.237.219 for hidden service 4m3v3i3bbix4aekoh5h2orlvtzwelnv4jqegocsjvyl46mwk4rfzpcqd
Aug 24 05:32:49.000 [warn] Failed to launch rendezvous circuit to $438F3EA4C9FB0DB63F5377A3271AB5435FAD7E04~ at 193.108.117.103
Aug 24 05:32:50.000 [warn] Giving up on launching a rendezvous circuit to $8838B9E8F70E10B5895427D370A65B5554C3F25D~ at 178.67.166.14 for hidden service 4m3v3i3bbix4aekoh5h2orlvtzwelnv4jqegocsjvyl46mwk4rfzpcqd
```

rootzoll, 2020-08-24:
OK so it seems that the Tor network itself is causing this unreliabilities .. there are two counter measures to consider:

1. On the bridge server it could reset to a new Tor identity if detected that there is a problem to reach then RaspiBlitz .. but that would maybe cut also all other Tor forwarding happening on that bridge.

2. The rendezvous server that is tied to the torv3 address might be under attack (because of another service listed there) and the RaspiBlitz could try to change the torv3 address address (delete old and let regenerate new) to be assigned to a different rendezvous server.

Before autimating the second option we could for the v1.6.1 build in a "detection" in the Tor notice logs to see if "Giving up on launching a rendezvous circuit" errors are happeing for a IP2TOR hidden service and and display this in service status information and also as part of the debug services. Also an CLI option to delete the IP2TOR sub plus refreshing the Torv3 address would be great so that the user can trigger that manually.

Also in general this makes the priority of also offering SSH-Reverse-Tunnels as an alternative subscription for people that like to trade more reliability for less anon.

rootzoll, 2020-09-16:
Also a possible fix would be: Setting up multiple hidden services (different rendezvous server) doing the same forwarding and having the bridge in a HAProxy kind of style always using the "working" connections.

But I think thats all for a future release to determine. Pushing this issue to v1.7 to take a look again.

openoms, 2020-09-16:
I don't think there is an issue with not using `SafeLogging=1` . 
We run a dedicated server which has much more things making possible to deanonymize the user if stolen for example.
On the other hand no apps should be able to peak into the logs without proper permissions.

closed 2022-12-13 by rootzoll