raspiblitz/raspiblitz #1450: Integrate Tallycoin Connect

rootzoll, 2020-08-13:
Connect a https://tallyco.in donation page directly to your RaspiBlitz :D
https://github.com/djbooth007/tallycoin_connect

nyxnor, 2021-04-27:
Even if running with `sudo python3 setup.py`, it doesn work because need to put sudo inside the variables to cp, enable, start. I did not manage to get the API key yet, still dont know what Im doing wrong.

rootzoll, 2021-04-28:
tanks for giving this a try - ca you connect with @djbooth007 to see whats still missing?

djbooth007, 2021-04-28:
> I did not manage to get the API key yet, still dont know what Im doing wrong on the last steps.

The API key can be generated in the admin dashboard after logging in to https://tallyco.in
Go to Node Connect section and copy the API key. Then, paste it in Tallycoin Connect on your Raspiblitz.
You can also check that your node is synced to the Tallycoin server (give it upto a minute).

> need to put sudo inside the variables to cp, enable, start.

Thanks, will update.

rootzoll, 2021-04-29:
@nyxnor whats the status on your integration test?

nyxnor, 2021-04-29:
Same as before. Im relating in this issue here https://github.com/djbooth007/tallycoin_connect/issues/3

nyxnor, 2021-04-30:
Ok. Im reporting  here now because I passed that problem with some changes he did.
Now I need to allow ufw. Am I doing something wrong?
```
admin@raspberrypi:~/tallycoin_connect $ sudo python3 setup.py
Created symlink /etc/systemd/system/multi-user.target.wants/tallycoin_connect.service → /etc/systemd/system/tallycoin_connect.service.
Enter your API key at http://127.0.1.1:8123/

sudo ufw allow 8123 comment 'tallycoin HTTP'
sudo ufw status

To                     Action      From
--                     ------      ----
8123                   ALLOW       Anywhere                 # tallycoin HTTP
8123 (v6)              ALLOW       Anywhere (v6)            # tallycoin HTTP
```
[image]

rootzoll, 2021-04-30:
`127.0.1.1` looks a bit strange  - maybe just a typo.

Firewall basically looks good ... are you sure that the service is running and serving on that port locally on the RaspiBlitz? 
What do you getting when you run a `wget 127.0.0.1:8123` or `wget 127.0.1.1:8123` on the terminal?

nyxnor, 2021-05-24:
I couldnt get this to work on my tests, but @pretyflaco did. You said you just followed the instruction on the tallycoin repo, but if you can just make a small install script to get this thing going.

rootzoll, 2021-12-21:
PR merged for final testing

rootzoll, 2021-12-21:
@dennisreimann I see the basic website but login is failing. In logs I see:
```
Dec 21 14:48:00 testlnd systemd[1]: Started Tallycoin Connect daemon.
Dec 21 14:54:35 testlnd npm[241392]: Error: Cannot find module '../encodings'
Dec 21 14:54:35 testlnd npm[241392]: Require stack:
```

Further TODOS:
- [ ] add as LND Service as one click install

dennisreimann, 2021-12-22:
Hmm, I just checked, for me it is working with Tallycoin Connect v1.7.0 and v1.7.1 (see #2835).

Tested on my RaspiBlitz v1.7.1 (Runtime = Node 14.17.6, npm 7.24.2)

dennisreimann, 2021-12-23:
@rootzoll If this still occurs for you, what are the Node and npm versions on RaspiBlitz v1.7.2?

rootzoll, 2022-01-04:
TODO: Restest with updated NodeJS see #2840

rootzoll, 2022-01-26:
OK sweet ... its finally running :D Will be part of v1.7.2
@dennisreimann thanks for hanging in there and putting the work on it

dennisreimann, 2022-01-26:
Nice, we're looking forward to the release and the feedback we get for our early version of the app. 🚀

rootzoll, 2022-02-09:
Just an observation to check further - on install during provision I got the report:

```
19067-nginx: [emerg] a duplicate listen [::1]:8125 in /etc/nginx/sites-enabled/tallycoin_connect_tor.conf:5
19068:nginx: configuration file /etc/nginx/nginx.conf test failed
19069:Job for nginx.service failed.
```

Also to check with `sudo nginx -t` after install.

dennisreimann, 2022-02-09:
@rootzoll Took a look and the port 8125 is only used for the Tallycoin Connect Tor service.

The other services configure the IPv6 listen without the `1`, so maybe we should adapt the nginx config for the tor and tor_ssl snippets like this:

```diff
-listen [::1]:8125;
+listen [::]:8125;
```

rootzoll, 2022-02-10:
OK looking good now :)

rootzoll, 2022-02-18:
OK install & recovery checked again - working. Closing for v1.7.2 release.

PR #2617 (2021-12-21): Integrate Tallycoin Connect
Closes #1450
closed 2022-02-18 by rootzoll