raspiblitz/raspiblitz #1453: ZAP on Android v1.6

byob75, 2020-08-14:
I am having trouble connecting to lnd from ZAP over TOR.

QR code works correctly, although takes a long time compared to Zeus.

Zeus on android works for me.

I noticed this port recently changed to 10009

rootzoll, 2020-08-14:
What is the error message of you trying to connect?

We were explicitly asked to changed to port 10009 for ZAP oder TOR - see:
https://github.com/rootzoll/raspiblitz/issues/1308#issue-649123975

please coordinate with @michaelWuensch if there is anything to change.

byob75, 2020-08-14:
Sorry I did not include before.

The error is:

_Unable to connect to LND
-make sure the LND daemon is running on your server.
-make sure your port forwarding is active.  (Port: 10009)
-make sure your certificate is valid_

I was not thinking I would have to do the port forwarding since I am using TOR onion address
It is probably my certificate, since I haven't done anything with a certificate yet.
Zeus complained about not having the certificate, but I bypassed, planning to go clean that up in the future.

ChuckNorrison, 2020-08-16:
i have documented the certificate stuff together with qr code and ip rather than name here: https://github.com/rootzoll/raspiblitz/issues/584

michaelWuensch, 2020-08-16:
Hi,
I finally did a tor setup with raspiblitz myself and then tried to connect with Zap Android & Orbot.
I used Raspiblitz 1.6.
Out of the box it did not work. The problem is that Zap Android does hostname verification which fails as the tls certificate does not include the tor hostname. After I manually added the tor hostname as tlsextradomain in lndconf and regenerated certificates it worked.
I will also update Zap Android next week to have better error messages to make finding the issue easier in the future.

So one solution would be that raspiblitz adds the tor hostname to the certificate.
Another option would be to remove hostname verification for tor connections on Zap Android.
I have heard someone say that there is no need to verify the tls certificate at all when connecting over tor, but I have very little experience with tor. If it is true though, then this would be the easiest solution.
Do you guys know if this would be safe?

rootzoll, 2020-08-17:
@openoms & @frennkie would it make sense (or hurt) to put the Tor hostname into the LND tlsextradomain configuration?

openoms, 2020-08-17:
> @openoms & @frennkie would it make sense (or hurt) to put the Tor hostname into the LND tlsextradomain configuration?

Don't think it would have any privacy implications since the tor address is advertised together with the public key.
What should not happen is to have dyndns or publicIP in there any time when running on Tor since that would give away the public IP to anyone connecting.

michaelWuensch, 2020-08-17:
@rootzoll @openoms  Actually my dyndns hostname is still in the lnd.conf and therefore in the certificate after I activated tor. So this gets leaked on connection over tor.

@openoms Do you know if it is neccessary for an app like Zap that is connecting to a lnd node behind tor to verify the certificate? Or can we omit that step as tor already makes sure we are communicating with the correct host?

michaelWuensch, 2020-08-17:
@openoms Do you know if it is necessary for an app like Zap that is connecting to a lnd node behind tor to verify the certificate? Or can we omit that step as tor already makes sure we are communicating with the correct host?

openoms, 2020-08-17:
SSL over Tor is not necessary, but since Tor is not built in the app it cannot separate the behaviour from going through clearnet.

I do recommend the Zeus wallet to people struggling with Zap. It has more active ongoing development and also an optional SSL cert verification with help in the app as well.
Check zeusln.app.

ChuckNorrison, 2020-08-17:
Im very happy with the zap app after the tls thing. Zeuss is still not working for me.

openoms, 2020-08-18:
> What should not happen is to have dyndns or publicIP in there any time when running on Tor since that would give away the public IP to anyone connecting.

Just to clarify this when switching to Tor there is a warning displayed recommending to reset the `lnd` wallet for privacy since the pubkey is already advertised together with the Public IP /  dyndns. 
Resetting the lnd wallet results in the deletion of the `lnd.conf` so creating a new wallet behind Tor does not result in advertising the old IP / dyndns together with the Tor address.
If the user does not reset their `lnd` wallet after swithcing to Tor (and they were running on clearnet before) they would need to delete the publicIP / dyndns entry manually from `lnd. conf`. This has no significant implications since that information was already the part of the network gossip.

michaelWuensch, 2020-08-19:
Hi,
I have now made a PR for Zap Android which removes certificate validation for tor connections. (https://github.com/LN-Zap/zap-android/pull/242)
A release with this change will most likely go live this or next week.

For Raspiblitz this means the tor hostname does not have to be added to the lnd.conf.
It also means that for tor connections lndconnect can be executed with --nocert  so we have a small nice QR-Code again.

Sorry for the trouble. But now I have setup it myself and can confirm this is working.

byob75, 2020-08-19:
> Hi,
> I finally did a tor setup with raspiblitz myself and then tried to connect with Zap Android & Orbot.
> I used Raspiblitz 1.6.
> Out of the box it did not work. The problem is that Zap Android does [...]

Hello,
 
when you add the lnd *.onion address from the bottom of the raspiblitz info screen to the /home/bitcoin/.lnd/lnd.conf, do you also add the :9735 port at the end?
tlsextradomain=0......yd.onion:9735

I tried with and without.

I then deleted the tls.cert and tls.key
and restarted the lnd service
I installed the new cert on an android phone

I was still not getting this to work correctly
even with forwarding ports 10009 and 9735

am I missing something?

Thank you

michaelWuensch, 2020-08-20:
> 
> @michaelWuensch
> 
> when you add the lnd *.onion address from the bottom of the raspiblitz info screen to the /home/bitcoin/.lnd/lnd.conf, do you also add the :9735 port at the end?
> tlsextradomain=0......yd.onion:9735
> 
> I tried with and without.
> 
> I then deleted the tls.cert and tls.key
> and restarted the lnd service
> I installed the new cert on an android phone
> 
> I was still not getting ZAP to work correctly (still getting Unable to connect to LND.... error msg)
> even with forwarding ports 10009 and 9735
> 
> am I missing something?
> 
> Thank you

Hey, definetly without port.
You need to use the onion address that is part of your lndconnect string. Then it should work.
What the Raspiblitz displays is the node Uri which is of the format: pubkey@host:port
To be honest I am surprised myself to see that the host part of of the node uri is different to the host included in the lndconnect string.
Can someone shed some light in this?

ChuckNorrison, 2020-08-20:
create a node uri an respect TOR or DNS is a problem in each project except ZAP. The ZAP App remembers my DynDNS Name and i never have to update any IP manually. Thats really great. But the TLS dependency is a problem to TOR support.

byob75, 2020-08-20:
> Hey, definetly without port.
> You need to use the onion address that is part of your lndconnect string. Then it should work.
> What the Raspiblitz displays is the node Uri which is of the format: [...]

@michaelWuensch

thanks so much!  

When I use the lndconnect string, I am now able to connect with ZAP :)

michaelWuensch, 2020-08-21:
Glad to hear it worked.
I just released a new version for Zap Android which removes the certificate verification for tor connections.
So tor issues should hopefully be a thing of the past now! It will be rolled out as soon as the appstore approval is finished.

rootzoll, 2020-08-24:
@michaelWuensch thanks for Zap Android update.

Reopening as a reminder to test this again for v1.6.1 release.

Marcoblue, 2020-09-28:
I set up Zap on a patched Raspiblitz 1.6 over TOR yesterday. Everything worked fine.

Only with Zeus there were problems, because QR Code is much too big.

rootzoll, 2020-10-12:
OK retested for v1.6.1 release ... was able to connect Zap Android over local IP & Tor (with Orbot). Closing issue.

closed 2020-10-12 by rootzoll