raspiblitz/raspiblitz #1455: Electrs only connecting over port 50001

inpharmaticist, 2020-08-15:
I was previously able to connect Electrum to electrs by the following methods

>electrum --oneserver --server 192.168.0.5:50002:s

>electrum --oneserver --server 192.168.0.5:50001:t

>electrum --oneserver --server [address].onion:50002:s --proxy socks5:127.0.0.1:9150

Now only the second one works. This is what the terminal shows using port 50002
[image]

This is using TOR
[image]

I have tried restarting nginx

>sudo systemctl restart nginx

As well as removing nginx, rerunning the ssl cert script, and rebooting (this use to get it working again but now it doesnt)
>sudo apt remove --purge nginx
>./config.scripts/internet.selfsignedcert.sh

Version 1.6

openoms, 2020-08-15:
If after a reinstall Electrum would not connect to Electrs delete the relevant old certificate on the desktop from:
    * Linux: ~/.electrum/certs/
    * Windows: User/appdata/roaming/electrum/certs/
More here: https://github.com/openoms/bitcoin-tutorials/blob/master/electrs/README.md#further-help

This is the last time you need to do this after an update. From 1.6 we preserve all the SSL certs on the disk.

inpharmaticist, 2020-08-15:
That worked, thanks!

closed 2020-08-15 by inpharmaticist