raspiblitz/raspiblitz #1457: Dropbox API

frennkie, 2020-08-15:
The Dropbox API that the Static Channel Backup uses is label as "Deprecated" on Dropbox apps page. It still works and I didn't find a defined date for when the old API will be switched of. But it might make sense to prepare for that.

[image]

rootzoll, 2020-08-17:
I am putting this on the v1.7 milestone for now to revisit it for the next release.

CandleHater, 2020-08-27:
I find it kind of odd to have an option like Dropbox in such a great FOSS project. In case you guys would be up for a nicer alternative like encrypted backup by E-Mail, Syncthing, etc.. Let me know.

frennkie, 2020-08-27:
I absolutely get your point..! Btw, (S/MIME) encrypted email is already available on Raspiblitz..! But as far as I recall it currently does not support attachments...

rootzoll, 2020-09-02:
Having DropBox fading out the old API is a good point for a switch to a better solution. The "Umbrel" people will also provide a backup store for SCB files in the future ... and all you need to access the file from their service automatically in case of a recovery are the seed words. So I think it can make sense to offer a "subscription" in the future for a similar SCB store (replacing DropBox) with an open source backend. This can be a fulmo subscription by default that people then can exchange with their own or other services.

shawnyeager, 2020-09-13:
Just attempted to setup the Dropbox backup with the above caveats, but it's failing. I get a glimpse of error messages, then I'm quickly back to the menu. If someone can point me to how to retrieve the correct logs, I'll post. Thanks.

rootzoll, 2020-09-14:
@shawnyeager the dropbox integration is running fine on my v1.6 release ... but added for retest as part of the v1.6.1 release. If you can get any details on the error message, that would help to get to the bottom of this.

shawnyeager, 2020-09-14:
> @shawnyeager the dropbox integration is running fine on my v1.6 release ... but added for retest as part of the v1.6.1 release. If you can get any details on the error message, that would help to get to the bottom of this.

@rootzoll Will do. Could you help me with where to pull logs?

Marcoblue, 2020-09-20:
Same here, I run a fresh Raspiblitz v1.6 and it never synced to Dropbox. If you tell me what you need, I will try to send it.

frennkie, 2020-09-20:
Does it help to look into the logs of the background service which is preforming the backup&upload?

```
sudo journalctl -f -u background -n 10000 | grep -i drop
```

Make sure to NOT post the output without hiding/masking the credentials!

> Also the "Download Backup" in RTL is not working. Code -2 ENOENT

I opened #1570 for that.

Marcoblue, 2020-09-20:
`sudo journalctl -f -u background -n 10000 | grep -i drop`
It shows nothing.

rootzoll, 2020-10-08:
it needs to be a `sudo journalctl -u background -b --no-pager -n 10000 | grep -i drop` ... the without `--no-pager` the journalctl never returns

rootzoll, 2020-10-17:
background has no output at all? Can you check if that service is running with `sudo systemctl status background`

Marcoblue, 2020-10-23:
No output at grep.
When I add the access token at Raspi, it says (only for short time): Error in call to Api function. Your app is not permitted to access this endpoint because it does not have the required scope filescontentwrite. The owner of the app can enable...

Than I enabled the files.content.write at Dropbox (also every other .write). When I insert the access token again, now it says: 
Dropbox Settings changed...
# Unknown error
upload=0
err='unknown'
errMode=#errorsummery missingscope error tag missingscope requiredscope filescontentwrite

Also the GUI at Dropbox changed. It differs to the readme.

rootzoll, 2020-10-26:
Maybe I need to test with a fresh Dropbox Account - will retest for v1.6.2 release.

rootzoll, 2020-10-30:
OK merged PR #1735 and test worked with a fresh dropbox account. I keep this issue open a bit for further test reports. But if nothing further pops up this will be part of v1.6.2 release and I am closing issue.

rootzoll, 2020-11-24:
Looks good on tsting - closing for v1.6.2 release.

closed 2020-11-24 by rootzoll