raspiblitz/raspiblitz #1458: Cancelling autounlock from menu fails [v1.6]

shawnyeager, 2020-08-17:
I was looking through the config options in menu, selected lnd autounlock, then, after being prompted to store the password on-device, selected `cancel`. The process continued anyway, confirming that autounlock is now active.

rootzoll, 2020-08-17:
Good find - will check the "CANEL" option on that dialog for the next v1.6.1 service release.

ChuckNorrison, 2020-08-24:
cancel button in reboot dialogue will also result in a reboot and not cancel

rootzoll, 2020-09-15:
@ChuckNorrison checked the reboot dialog on SSH main menu and it respects my cancel.
@shawnyeager found the problem in the auto-unlock cancel. Will be fixed in v1.6.1

rootzoll, 2020-10-11:
OK Testing works. Will be part of v1.6.1 release .. closing issue.

closed 2020-10-11 by rootzoll