raspiblitz/raspiblitz #1461: Lnd not starting after updating to v1.6

syracusebitcoin, 2020-08-17:
Hey i just updated to V1.6 and Lnd is not starting. Here is the debug log if anyone might be able to help me figure out what i messed up. please and thank you.

blitzversion: 1.6.0
chainnetwork: bitcoin / main
 23:26:29 up 12 min,  3 users,  load average: 1.03, 1.40, 0.95

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-08-17 23:23:03 BST; 3min 26s ago
  Process: 11192 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 11194 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 11195 (bitcoind)
    Tasks: 18 (limit: 4915)
   Memory: 256.3M
   CGroup: /system.slice/bitcoind.service
           └─11195 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitco…

Aug 17 23:23:03 raspberrypi systemd[1]: bitcoind.service: Can't open PID …tory
Aug 17 23:23:03 raspberrypi systemd[1]: Started Bitcoin daemon.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Mon 2020-08-17 23:13:40 BST, end at Mon 2020-08-17 23:26:29 BST. --
Aug 17 23:18:11 raspberrypi systemd[1]: Starting Bitcoin daemon...
Aug 17 23:18:11 raspberrypi systemd[1]: Started Bitcoin daemon.
Aug 17 23:22:46 raspberrypi systemd[1]: Stopping Bitcoin daemon...
Aug 17 23:22:48 raspberrypi systemd[1]: bitcoind.service: Succeeded.
Aug 17 23:22:48 raspberrypi systemd[1]: Stopped Bitcoin daemon.
Aug 17 23:23:02 raspberrypi systemd[1]: Starting Bitcoin daemon...
Aug 17 23:23:03 raspberrypi systemd[1]: bitcoind.service: Can't open PID file /home/bitcoin/.bitcoin/bitcoind.pid (yet?) after start: No such file or directory
Aug 17 23:23:03 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-08-17T22:26:19Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:20Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:20Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:21Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:21Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:22Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:22Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:23Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:23Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:24Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:24Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:25Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:25Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:26Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:26Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:27Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:27Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:28Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:28Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)
2020-08-17T22:26:29Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2020-08-17 23:26:04 BST; 24s ago
  Process: 12929 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
  Process: 12931 ExecStart=/usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --listen=127.0.0.1:9735 ${lndExtraParameter} (code=exited, status=1/FAILURE)
 Main PID: 12931 (code=exited, status=1/FAILURE)

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Mon 2020-08-17 23:13:40 BST, end at Mon 2020-08-17 23:26:29 BST. --
Aug 17 23:25:03 raspberrypi systemd[1]: Started LND Lightning Daemon.
Aug 17 23:25:04 raspberrypi lnd[11910]: dial tcp 127.0.0.1:9050: connect: connection refused
[... 186 lines cut ...]
raspberryPi=4
powerFAIL=0
powerWARN=0
powerMIN='862500 microVolt'
tempFAIL=11
tempWARN=15
tempMAX='8000 centiGrad'
IMPORTANT: There are some hardware issues with your setup.
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.0.13'
tempCelsius='72.0'
uptime=761
upsStatus='OFF'
startcountBlockchain=3
bitcoinActive=1
blockchainHeight=644184
initialSync=0
syncProgress=100.00 
startcountLightning=11
lndActive=0
blitzTUIActive=1
blitzTUIRestarts=0
scriptRuntime=1

ChuckNorrison, 2020-08-18:
this system lost a lot of stuff in this upgrade.

> sudo: nginx: command not found

Someone who got some backup experience should check this out

rootzoll, 2020-08-18:
@openoms and @frennkie I start to see a pattern here ... a lot of bug reports lately have todo with "installs not running correctly" lets talk next week how we can improve this maybe thru pre-packaging apt-get, pip and npm depencendies on the sd card - this would also make the RaspiBlitz better suitable for areas with flaky, costly or low-bandwidth internet.

@syracusebitcoin if possible send me a copy of your `/home/admin/raspiblitz.recover.log` to christian@fulmo.org and then try an update again with a fresh sd card that hopefully this time all dependecies get loaded and installed correctly.

syracusebitcoin, 2020-08-29:
Hello everyone. Thanks for the help. @rootzoll the problem was indeed a bad SD card. I recently got a new one, re-flashed the img and it seems to be working fine for the most part. There seem to be a few kinks here and there but I think with a little bit of research I can fix these issues. Again thanks everyone for the help.

closed 2020-08-29 by syracusebitcoin