raspiblitz/raspiblitz #1462: Blockchain started syncing then got 'Service bitcoind is not running' message

kbholden, 2020-08-18:
Debug report for 'Service bitcoind is not running' message

admin@raspberrypi:~ $ /home/admin/XXdebugLogs.sh | nc termbin.com 9999
tail: error reading '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log': Input/output error
nginx: [emerg] BIO_new_file("/mnt/hdd/app-data/nginx/tls.cert") failed (SSL: error:02001005:system library:fopen:Input/output error:fopen('/mnt/hdd/app-data/nginx/tls.cert','r') error:2006D002:BIO routines:BIO_new_file:system lib)
nginx: configuration file /etc/nginx/nginx.conf test failed
/home/admin/config.scripts/blitz.statusscan.sh: line 4: /mnt/hdd/raspiblitz.conf: Input/output error
/home/admin/config.scripts/blitz.ups.sh: line 4: /mnt/hdd/raspiblitz.conf: Input/output error
https://termbin.com/1u3g

ChuckNorrison, 2020-08-18:
seems your nginx is missing his tls.cert? you can try to repair in menu your certificates

rootzoll, 2020-08-24:
@kbholden bitcoind AND lnd do not run correctly. Something went bad during install. If this is a fresh install I would recommend to start fresh: Format HDD/SSD on laptop and start again with fresh sd card image. If this is after an update: Shutdown and reboot with a fresh sd card again.

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll