raspiblitz/raspiblitz #1464: DynamicDNS is not used at all and we should not take care now

ChuckNorrison, 2020-08-18:
Iam testing a lot with the lightning setup and all the projects are in. I observed that especially eclair and https://explorer.acinq.co is using publicIp only and the dont update them if changed. 

So on eclair the node ip needs to be updated manually for each ip change, whether or not the domain name was used for that configuration.

on the acinq explorer webinterface it never updates the ip. the first publicIP which was used by the node is displayed there and nobody can open a channel with.

next the ssl thing takes place and makes everything more weird.

So the only setting which will work without using tor is a VPS with static ip+VPN setup to route the lighning traffic where you want. Any recommendations on this?

ChuckNorrison, 2020-08-18:
The best written documentation on this with OpenVPN in detail found here:
https://www.digitalocean.com/community/tutorials/how-to-set-up-an-openvpn-server-on-ubuntu-18-04

If i setup openvpn client in that RaspiBlitz in background, what steps should i be aware of? In my opinion, the raspi and admin scripts should not care if VPN tunnel is fully setup. Even the LetsEncrypt + DynamicDNS Feature should still do his job. The client part should not be a big task to setup. The openvpn server can be tricky on first time.

ChuckNorrison, 2020-08-20:
I did all the steps from this awesome OpenVPN Tutorial. Everything works great, my node gets the static public ip i registered on a VPS. I will close this, but OpenVPN is a big tool, maybe the blitz can deliver some ovpn import possibilities

closed 2020-08-20 by ChuckNorrison

rootzoll, 2020-08-20:
What benefit do you see in using OpenVPN? There is the feature already of using reserve SSH forwarding you can set with the RaspiBlitz: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-setup-port-forwarding-with-a-ssh-tunnel

ChuckNorrison, 2020-08-20:
i want to fight the QR code and DNS compatibility with a static public IP i dont get at my private ISP. So for additional security it was a long time ago i frist wanted to setup a VPN Gateway for mobile and some services i want to host on a static ip.

at raspiblitz i setup openvpn client service and at openvpn server i forward our common ports 5001:9735:10009. That seems to work very great. But the server side is a bit complex

ChuckNorrison, 2020-08-20:
this can be helpful on a blitz to only allow traffic via openvpn and internal ssh

`#!/bin/bash

# Reset firewall rules
sudo ufw reset

# deny traffic
sudo ufw default deny incoming
sudo ufw default allow outgoing

# allow ssh in every case
sudo ufw allow from 192.168.1.0/24 to any port 22 proto tcp

# allow OpenVPN Client connection
sudo ufw allow out 1194/udp

# allow traffic via openvpn tunnel only
sudo ufw allow out on tun0

# enable rules
sudo ufw enable`