raspiblitz/raspiblitz #1466: error code: -28 error message: Loading block index...

grnqrtr, 2020-08-19:
I get this on every reboot and and I've seen a number of other people get this on 1.6. Seems like it is giving up to early or something, doesn't give enough time to let the block index load.
```
*****************************************
* The bitcoind service is not running.
*****************************************
If you just started some config/setup, this might be OK.

More Error Detail:
error code: -28 error message: Loading block index...

POSSIBLE OPTIONS:
-> Use command 'repair' and then choose 'DELETE-INDEX' to try rebuilding transaction index.
-> Use command 'repair' and then choose 'RESET-CHAIN' to try downloading new blockchain.
-> Use command 'debug' for more log output you can use for getting support.
-> Use command 'menu' to open main menu.
-> Have you tried to turn it off and on again? Use command 'restart'

bitcoin error: error code: -28
error message:
Loading block index...
```

ChuckNorrison, 2020-08-19:
hey i get this too after power off and on. last reboot it worked. Everytime if that happens, i have to update public ip manually with the dynupdate url

Jestopher-BTC, 2020-08-19:
Just got this error with a completely fresh SD.

bitsam, 2020-08-20:
run, sudo tail -n 20 /mnt/hdd/bitcoin/debug.log

See if it is slowly verifying the blocks or downloading blocks. Once the chain is sync'd then you can run , raspiblitz to launch the menu and go back through the normal loading without a reboot. 

I also find running the lnd wallet unlock to help with the command, lncli unlock

bitsam, 2020-08-20:
I had some teething problems with 1.6, and decided to blow away the chain data. Which was probably an over-reaction, as once I had finally copied it over I still got the above errors, and it was caused by the the slow verification of blocks or need to download the last few remaining blocks. You can check all this in the /bitcoin/debug.log

ChuckNorrison, 2020-08-20:
so its just a race condition between the bitcoind and lnd. Wait for start lnd while bitcoind is syncing. Not that easy for a script to get that state i think. i depends a lot of sync time... than can be days or weeks. At setup from scratch that all worked well.

rootzoll, 2020-08-24:
This seems to be a timing issue .. marked for v1.6.1 to change timing on this.

thecharacter, 2020-08-30:
I figured out a workaround if you type menu in the Commandline then select Info from the menu options it will ask you to type in your LND password and then the load index window will appear letting you know what percentage your Blockchain Index and LND index is at.

rootzoll, 2020-09-02:
merged the fix by @CandleHater - putting it on final testing

rootzoll, 2020-09-03:
original fix by @CandleHater tested good .. rolling with that.

rootzoll, 2020-10-08:
Tested for some weeks now ... looks goof. Will be part of v1.6.1. Closing this issue.

closed 2020-10-08 by rootzoll