raspiblitz/raspiblitz #1467: bitcoind service not running error on each reboot

Jestopher-BTC, 2020-08-19:
Every time I restart my node, I get prompted with the following error. After a few moments , I get notified that my raspiblitz is running. When I enter 'raspiblitz' instead of debugging, everything works as expected.

*****************************************
* The bitcoind service is not running.
*****************************************
If you just started some config/setup, this might be OK.

More Error Detail:
error code: -28 error message: Loading block index...

POSSIBLE OPTIONS:
-> Use command 'repair' and then choose 'DELETE-INDEX' to try rebuilding transaction index.
-> Use command 'repair' and then choose 'RESET-CHAIN' to try downloading new blockchain.
-> Use command 'debug' for more log output you can use for getting support.
-> Use command 'menu' to open main menu.
-> Have you tried to turn it off and on again? Use command 'restart'

bitcoin error: error code: -28
error message:
Loading block index...

Debug log for reference:
https://termbin.com/spge4

bitsam, 2020-08-19:
Looking at the log, the bitcoind service is still verifying the blocks between 16-32%. You can keep an eye on this by re-running the command. sudo tail -n 20 /mnt/hdd/bitcoin/debug.log

and see if the progress increases. I also notice that things get a little out of order and sometime the system wants you to unlock your wallet. so you can also force this via the command prompt with the following command, lncli unlock

I would let it run for a while and keep and eye on the bitcoin/debug.log and see if it progresses, it may also have to download several blocks if you have been offline for a while. BTW, were you running 1.6 for a while before this happened, or did this happen immediately after upgrading to v1.6?

hermanlim, 2020-08-20:
Try waiting a while (5-10 mins should be sufficient) and then running raspiblitz from command line.

rootzoll, 2020-08-24:
This might be a timing issue also @CandleHater reported .. normally waiting like @hermanlim mentioned should be a workaround for now. Will take a look to see if we can optimze here for the v1.6.1 release.

CandleHater, 2020-08-25:
I guess a good solution to this is checking if `bitcoinErrorFull` has Error -28 (means RPC_IN_WARMUP, see https://github.com/bitcoin/bitcoin/blob/master/src/rpc/protocol.h#L49) around here https://github.com/rootzoll/raspiblitz/blob/aef8b3a1da5c0df5916ba828f5adc808fd2de624/home.admin/80scanLND.sh#L47

thecharacter, 2020-08-30:
I figured out a workaround if you type menu in the Commandline then select Info from the menu options it will ask you to type in your LND password and then the load index window will appear letting you know what percentage your Blockchain Index and LND index is at.

rootzoll, 2020-09-02:
closing this issue ... will be fixed as part of #1466 for v1.6.1 service release.

closed 2020-09-02 by rootzoll