raspiblitz/raspiblitz #1470: Running the numbers .. validate show how many Bitcoins are issued

rootzoll, 2020-08-20:
Either do it once this is merged into bitcoind: https://github.com/bitcoin/bitcoin/pull/19521 or do it the manual way.

CandleHater, 2020-08-20:
LMAO there we go

rootzoll, 2020-09-28:
Can be done easily by Specter 0.8 that will be part of the v1.6.1 release:
https://twitter.com/KevinR4v/status/1310583241942990848?s=20

But maybe also having that part of the LCD infoscreen would be nice :)

openoms, 2020-09-28:
It is a quite CPU intense to be just run automatically imo.
Specter integrates it nicely or can be called from the cli any time.

rootzoll, 2020-09-28:
> It is a quite CPU intense to be just run automatically imo.

if `blockfilterindex=1` in `bitcoin.conf`is set ... ist that just a small update per new block?
@fjahr how intensive is that for a RaspberryPi to regularly poll `gettxoutsetinfo` with `blockfilterindex=1`? Too much load?

openoms, 2020-09-28:
The `blockfilterindex=1` is to do with the BIP158 (client-side-filtering - Neutrino) block filters, not the UTXO set.
The coinstat features which will include ongoing low resource `gettxoutsetinfo` monitoring is still WIP. Just listened: https://stephanlivera.com/episode/214/ ;)

fjahr, 2020-09-28:
> The `blockfilterindex=1` is to do with the BIP158 (client-side-filtering - Neutrino) block filters, not the UTXO set.
> The coinstat features which will include ongoing low resource `gettxoutsetinfo` monitoring is still WIP. Just listened: https://stephanlivera.com/episode/214/ ;)

Correct :) It's still going to need some time until https://github.com/bitcoin/bitcoin/pull/19521 is going to be merged I guess. 

When it's integrated it's only very little additional load, should be about 1 second to update the index on a slow machine after a block has been validated but I haven't benchmarked it lately. Then getting the values from `gettxoutsetinfo` just reads the records from the index. The change is very much aimed at making it possible to enable continuous polling of gettxoutsetinfo on a low powered machine like RPi3. Only having the index catch up initially will take a couple of hours.

rootzoll, 2020-09-28:
Ah OK .. mixed those indexes up :) So we will monitor if https://github.com/bitcoin/bitcoin/pull/19521 gets merged and then consider to add continuous polling for displaying to the user. @fjahr thanks for the work

rootzoll, 2021-04-05:
This is already part of pyblock and also possible within specter. So I think we can close this one.

closed 2021-04-05 by rootzoll