raspiblitz/raspiblitz #1485: Tor info.log growing endlessly

openoms, 2020-08-24:
Had two occasions with BTRFS disks when the 30GB `sda1` (`/mnt/hdd`) has clogged up with a 29GB `info.log` file in /mnt/hdd/tor/.

Need to test if it growing similar in the standard EXT partitions.

To check:
`sudo ls -la /mnt/hdd/tor`

remove with:
`sudo rm -rf /mnt/hdd/tor/info.log`
and restart the node to delete the file.

Need to find a way to prune the log file or delete it occasionally.

rootzoll, 2020-08-24:
Quickfix for 1.6.1 -> delete on bootup
Find better solution for 1.7 when Tor gets default.

frennkie, 2020-08-24:
How about logrotate..?! ;-D

rootzoll, 2020-08-24:
@frennkie is there a logrotate option in tor or is this an external setting?

frennkie, 2020-08-24:
Logrotate is shipped with Raspbian.. there even is a Tor config.. it just doesn't include our modified directory:

```
$ cat /etc/logrotate.d/tor
/var/log/tor/*log {
        daily
        rotate 5
        compress
        delaycompress
        missingok
        notifempty
        create 0640 debian-tor adm
        sharedscripts
        postrotate
                if invoke-rc.d tor status > /dev/null; then
                        invoke-rc.d tor reload > /dev/null
                fi
        endscript
}

```

I guess all that is needed it to add a line `/mnt/hdd/tor/*.log` to the top.

frennkie, 2020-09-05:
My info.log was at 37 GB today.. /o\ opened PR to fix

rootzoll, 2020-09-08:
Merged PR ... final testing for v1.6.1 service release.

rootzoll, 2020-09-10:
also test if the #1506 merge works fine

rootzoll, 2020-09-28:
also test if the #1571 merge works fine

rootzoll, 2020-09-28:
Problem needs to be resolved: https://github.com/rootzoll/raspiblitz/pull/1571#issuecomment-700282106

rootzoll, 2020-10-08:
OK running this now for some days and looks good so far. Will be part of v1.6.1 - closing issue.

PR #1507 (2020-09-08): setup logrotate for Tor on ext. disk
fix #1485
closed 2020-10-08 by rootzoll