raspiblitz/raspiblitz #1488: Update: JoinMarket v0.7.0 (with joininbox menu v1.3.1)

openoms, 2020-08-25:
working to refactor the update script
The challenge is to use system-site-packages in the jmvenv virtual env to keep the QT GUI working from the blitz.

rootzoll, 2020-09-02:
merged PR - setting for final testing

rootzoll, 2020-10-09:
Install on final testing is working. Will be part of v1.6.1 release. Closing issue.

openoms, 2020-10-13:
JoinMarket v0.7.1 is released, just testing installation. Could not find significant differences in dependencies.

openoms, 2020-10-14:
To update JoinMarket separately can use this much improved script:
```
# download from this repo
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.6/home.admin/config.scripts/bonus.joinmarket.sh
# uninstall old version (wallets are preserved on the disk)
sudo bash bonus.joinmarket.sh off
# install with the latest updates
sudo bash bonus.joinmarket.sh on
```

rootzoll, 2020-10-14:
OK install seems to work - rest is up to you @openoms to test on Release Candidate. Will be part of v1.6.1 - closing this issue.

closed 2020-10-14 by rootzoll