raspiblitz/raspiblitz #1491: How to add a second HDD?

Pinkman420, 2020-08-26:
Hello. I would like to connect a second HDD to my Raspiblitz on RPi4. Unfortunately, the Raspiblitz smears off immediately when I plug in the HDD and then doesn't boot anymore. Only a new installation of the whole system will help. I would like to use more than just one USB port ;-) The fstab remains unchanged so unfortunately I don't know what the problem is.

Migrating is not possible and I have to connect both HDDs to another Pi to synchronize the hard drives. So today a lot of downtime again ...

linuxxp, 2020-08-26:
Hello. You can use Clonezilla (Bootable) or Macrium Reflect (Windows) to clone the main HDD.

Pinkman420, 2020-08-26:
Thanks for the hint. I do this with rsync on another Raspberry. But in the future it should be possible to use more like just one USB port or more like just one HDD. I can hardly imagine that everyone here uses their Rpi exclusively for Raspiblitz.
In addition, I would still be very interested why nothing works as soon as I connect a second HDD and there is no other way except a complete new installation.

I connect the hard drives to my other pis, mount them and there are no problems. I connect a second HDD to Raspiblitz and nothing works, not even after removing the second HDD. Only a complete new installation will help.

rsync has been running for 3 hours and will need as much time again (I have more than just the blockchain stored on it), then the new installation, install all packages. So the node is down the complete day and that's only because I can't connect a second HDD to Raspiblitz. otherwise rsync could simply have run on the side without the node being down. This is a joke.

hermanlim, 2020-08-29:
There have been concerns voiced regarding SSD/HDD power consumption so a second device is not recommended, at least not without an external power source. Doing so risks data corruption and potential list of funds. 

This is a raspberry hardware issue.

Pinkman420, 2020-08-31:
Thanks for the answer.

Both hard drives are operated with external power supply, so it cannot be because of this.

If I operate these two hard drives with Raspian without Raspiblitz, it is not a problem, only with Raspiblitz. I run several HDDs on all my Pis, only on the Raspiblitz this is not possible.

hermanlim, 2020-08-31:
Have you previously used the same HDD cable successfully? RPi4 seems quite picky when it comes to the USB Sata cable.

I have had issues even within the same brand of HDD enclosure/Cable

See here:

https://www.raspberrypi.org/forums/viewtopic.php?t=245767

Pinkman420, 2020-09-01:
Yes, I just changed the SD card and ran Raspian without Raspiblitz because of rsync. There were no problems there. I also tested the hard drives on 2 other Pis, no problems there either.

In the Raspiblitz menu there is also the note that you should use a USB stick as a backup and not another hard drive.

So it has to have something to do with Raspiblitz. I just don't know why because the fstab remains unchanged.

What I haven't tested yet is a different cable. It would be worth a try, but the constant reinstallation is annoying and very time-consuming. I will then give you feedback again.

hermanlim, 2020-09-01:
Given you ran both drives without issue on Raspbian without issue makes it significantly less likely to be a cable/enclosure issue. The cable issue happens usually within 10 minutes.  Bugger, I was hoping to give you a solution! 

Please post an update when you find the solution.