raspiblitz/raspiblitz #1492: Joinmarket mixed ownership of scripts/*.py

nolith, 2020-08-26:
Hello, after installing joinmarket I noticed that I was not able to edit the yield-generator from the menu.

I got no errors, but opening the fail again gave me the old content.

Checking the content of the disk, I noticed that some files are owned by `root`, and `joinmarket` cannot edit them.

```
(jmvenv) joinmarket@raspberrypi:~/joinmarket-clientserver/scripts $ ls -l
total 244
-rw-r--r-- 1 joinmarket joinmarket  3311 Aug 25 07:26 README.md
drwxr-xr-x 2 joinmarket joinmarket  4096 Aug 25 07:33 __pycache__
-rwxr-xr-x 1 root       root        8968 Aug 25 07:26 add-utxo.py
drwxr-xr-x 2 joinmarket joinmarket  4096 Aug 25 07:26 cmtdata
-rw-r--r-- 1 joinmarket joinmarket  1234 Aug 26 06:26 commitmentlist
-rwxr-xr-x 1 root       root        5285 Aug 25 07:26 convert_old_wallet.py
-rwxr-xr-x 1 root       root       88412 Aug 25 07:26 joinmarket-qt.py
-rwxr-xr-x 1 joinmarket joinmarket   120 Aug 25 07:26 joinmarket-qt.sh
-rwxr-xr-x 1 joinmarket joinmarket  1368 Aug 25 07:26 joinmarketd.py
drwxr-xr-x 2 joinmarket joinmarket  4096 Aug 25 07:26 logs
drwxr-xr-x 3 joinmarket joinmarket  4096 Aug 25 07:34 obwatch
-rw-r--r-- 1 root       root       37189 Aug 25 07:26 qtsupport.py
-rwxr-xr-x 1 joinmarket joinmarket  3614 Aug 25 07:26 receive-payjoin.py
-rw-r--r-- 1 joinmarket joinmarket  1859 Aug 25 07:26 sample-schedule-for-testnet
-rwxr-xr-x 1 root       root       14714 Aug 25 07:26 sendpayment.py
-rwxr-xr-x 1 root       root        4758 Aug 25 07:26 sendtomany.py
-rwxr-xr-x 1 joinmarket joinmarket  8783 Aug 25 07:26 tumbler.py
-rwxr-xr-x 1 joinmarket joinmarket   168 Aug 25 07:26 wallet-tool.py
drwxr-xr-x 2 joinmarket joinmarket  4096 Aug 25 07:26 wallets
-rwxr-xr-x 1 root       root        4649 Aug 25 08:25 yg-privacyenhanced.py
-rwxr-xr-x 1 joinmarket joinmarket   593 Aug 25 07:26 yield-generator-basic.py
``` 

/cc @openoms

openoms, 2020-08-26:
Must be that the install was run with `root` (`sudo`) instead of the `joinmarket` user

Did you have a custom install or update by any chance?

This one maybe:
https://github.com/rootzoll/raspiblitz/pull/1489#issuecomment-679967237

openoms, 2020-08-26:
Hope the last xommit fixes this.
Can you try this to reinstall the new version:
```
sudo rm -f bonus.joinmarket.sh
wget https://raw.githubusercontent.com/openoms/raspiblitz/jm-0.7.0/home.admin/config.scripts/bonus.joinmarket.sh
chmod +x bonus.joinmarket.sh
sudo ./bonus.joinmarket.sh off
sudo ./bonus.joinmarket.sh on
```

nolith, 2020-08-26:
> Did you have a custom install or update by any chance?

Nope. Just a regular installation from the services menu.

> Can you try this to reinstall the new version:

It seems to be fixed now

```
 $ ls -l ~joinmarket/joinmarket-clientserver/scripts/
total 240
-rw-r--r-- 1 joinmarket joinmarket  3311 Aug 26 14:40 README.md
-rwxr-xr-x 1 joinmarket joinmarket  9032 Aug 26 14:40 add-utxo.py
drwxr-xr-x 2 joinmarket joinmarket  4096 Aug 26 14:40 cmtdata
-rwxr-xr-x 1 joinmarket joinmarket  5228 Aug 26 14:40 convert_old_wallet.py
-rwxr-xr-x 1 joinmarket joinmarket 93415 Aug 26 14:40 joinmarket-qt.py
-rwxr-xr-x 1 joinmarket joinmarket   120 Aug 26 14:40 joinmarket-qt.sh
-rwxr-xr-x 1 joinmarket joinmarket  1368 Aug 26 14:40 joinmarketd.py
drwxr-xr-x 2 joinmarket joinmarket  4096 Aug 26 14:40 logs
drwxr-xr-x 2 joinmarket joinmarket  4096 Aug 26 14:40 obwatch
-rw-r--r-- 1 joinmarket joinmarket 37163 Aug 26 14:40 qtsupport.py
-rwxr-xr-x 1 joinmarket joinmarket  3614 Aug 26 14:40 receive-payjoin.py
-rw-r--r-- 1 joinmarket joinmarket  1859 Aug 26 14:40 sample-schedule-for-testnet
-rwxr-xr-x 1 joinmarket joinmarket 16380 Aug 26 14:40 sendpayment.py
-rwxr-xr-x 1 joinmarket joinmarket  4951 Aug 26 14:40 sendtomany.py
-rwxr-xr-x 1 joinmarket joinmarket  8783 Aug 26 14:40 tumbler.py
-rwxr-xr-x 1 joinmarket joinmarket   168 Aug 26 14:40 wallet-tool.py
drwxr-xr-x 2 joinmarket joinmarket  4096 Aug 26 14:40 wallets
-rwxr-xr-x 1 joinmarket joinmarket  4614 Aug 26 14:40 yg-privacyenhanced.py
-rwxr-xr-x 1 joinmarket joinmarket   593 Aug 26 14:40 yield-generator-basic.py
```

Thank you

closed 2020-08-26 by nolith