raspiblitz/raspiblitz #1495: fail2ban ssh authentications based on systemd

ChuckNorrison, 2020-08-27:
After fail2ban and ubuntu gets updated, it needs to be set to backend=systemd to work as expected

`sudo nano jail.conf`

find backend and set from auto to systemd. 

`sudo systemctl restart fail2ban`

Now we can check for authentication failures on our ssh port with:

`sudo fail2ban-client status sshd`

ChuckNorrison, 2020-09-02:
1h is still great. For me its easy task to log from another machine to remove my ip ban manually, but for new users it could be a mess... maybe a whitelist entry from local network could be possible to avoid selfbans and only check for authentications from unknown

freerko, 2021-01-06:
stumbled upon this issue looking for the current fail2ban setting of raspiblitz because I tried a wrong password too often. Please don't increase it even further, 10min should be more than enough to prevent brute-force attacks via SSH. 
Adding private networks 10.0.0.0/8, 172.16.0.0/12 and 192.168.0.0/16 to the ignore list would be very helpful as well.

rootzoll, 2021-01-10:
I am not sure about whitelisting local networks ... there are too many smart devices nowadays in your local network. But yeah 10min should be enough ... I just recently banned myself by using wrong password on wrong machine. Even the 10min wait was bad enough ... and should be fine for practical reasons.

closed 2021-03-18 by rootzoll