raspiblitz/raspiblitz #1498: "The network connection was lost." when trying to connect LN Node to Fully Noded.

Fittiboy, 2020-08-27:
I successfully connected the Bitcoin node to Fully Noded through the main menu option. I would also like to use the app's lightning functionality, but when trying to connect the lightning node, I encounter a problem:
When adding the node to the app, I enter the following information:
Label: Some random name I chose
RPC username: The RPC username from bitcoin.conf (raspibolt)
RPC password: The RPC password from bitcoin.conf (password B)
Onion address: The onion address from the "Info" screen (everything after the "@")

Now even after restarting both the app and the RaspiBlitz, I get the error message "The network connection was lost." in the Fully Noded app whenever I try to use any lightning feature.
Not sure whether this is a RaspiBlitz issue or a Fully Noded issue.

rootzoll, 2020-09-02:
maybe check `sudo cat /etc/tor/torrc` to see if all the port forwardings needed on the hidden service are active

Fittiboy, 2020-09-10:
Tor seems to be set up correctly. I have no issues connecting when using the Zap iOS and desktop Apps.
Very weird...

rootzoll, 2020-09-10:
> Onion address: The onion address from the "Info" screen (everything after the "@")

I think you are using the wrong onion address.  The one from your NodeURI is just for the Lightning-P2P-Communication (protocol port 9735). You want to connect to RPC port (locally 10009). Those are running under a different onion-address. type `sudo cat /mnt/hdd/tor/lndrpc10009/hostname` on terminal to get the correct onion-address for the LND RPC port.

Fittiboy, 2020-09-10:
Trying that also didn't work. I did some research and found out that Fully Noded apparently [only supports c-lightning](https://github.com/Fonta1n3/FullyNoded/blob/0402d08013894b8d0bff1cfb539a191d0643a504/Docs/Q-and-A.md#can-i-turn-on-my-lnd-server-on-mynode-to-use-fn-as-a-lightning-wallet-in-conjunction-with-an-onchain-wallet).

closed 2020-09-14 by rootzoll