raspiblitz/raspiblitz #1504: raspiblitz installed on 8GB , but sdcard is 32GB

laurentdebricon, 2020-09-04:
Hello,
fresh install raspiblitz 1.6

today i ran out of space when installing btcpayserver, then i discovered that 

/dev/root       7.0G  6.6G     0 100% /

- i bought the 32G "recommended" sdcard (Samsung PRO Endurance 32 GB microSDHC UHS-I U1)
- flashed with balena on Ubuntu

what went wrong ?
how to fix, and expand the partition ?

openoms, 2020-09-04:
For some reason the SDcard was not expanded on first boot

Make an LND backup from the BACKUP menu
that stops all services as well
(press CTRL+C before the node restarts)
and run:
`sudo raspi-config --expand-rootfs`

Related:
https://github.com/rootzoll/raspiblitz/issues/936

laurentdebricon, 2020-09-05:
hey ! thank you !

sadly i had also kernel panic at boot ... so :

- i read the code of the lnd backup script on github, connect the HDD on my computer and did my little tar.gz by myself :)
- reflashed the SDcard
- rescue script kicked and succeeded. no need of the lnd.backup.tar.gz (but playing safe is always good)

wow i learned so much today, thanks !

closed 2020-09-05 by laurentdebricon