raspiblitz/raspiblitz #1515: LND service will not run after upgrade to 1.6

jalal, 2020-09-07:
I recently reflashed my sd card with 1.6 and restarted the Raspiblitz. However there seems to be no running `lnd` daemon and I cannot get one to startup. 
The system is a preloaded RP4 and came with 1.4, I upgraded to 1.5 with no problem and now decided to upgrade to 1.6 -- so far, no luck.

Here are some details:

Burn SD with new Raspiblitz 1.6
Insert and start.
Start up ok
$ ssh admin@192.168.0.99
Connection refused

Wait one hour, no disk activity, power down Pi, restart.

ssh is now ok

Enter a new password A
Confirm
"OK - password A was set for all users..."

Dialog stays, message at top left:
Enabling services
(wait 2 mins)
Failed to reload daemon: Connection timed out
(wait 3 mins)
Failed to enable unit: Connection timed out
(wait 3 mins)
Failed to enable unit: Connection timed out
(wait 5 minutes)
Dialog: "You had the Autounlock feature, enter password C..."

"# Testing password .. please wait"
"FINAL REBOOT IS NEEDED"
"LCD turns white...
It will then reboot again automatically
stop lnd - please wait ..
stop bitcoind (1) - please wait ..
stop bitcoind (2) - please wait ..
starting shutdown
"

No lnd daemon found

```
$ lncli getinfo
[lncli] open /home/admin/.lnd/tls.cert: no such file or directory
$ sudo systemctl status lnd
Failed to get properties: Connection timed out
$ sudo systemctl enable lnd
Failed to enable unit: Connection timed out
$ sudo systemctl start lnd
Failed to enable unit: Connection timed out
```
I get the feeling that the systemd is borked, but I've no idea what could cause that, nor how to fix it.

I've created a XXdebuglog at:

uniibu, 2020-09-07:
You might want to reinstall lnd.
First backup your lnd:
```/home/admin/config.scripts/lnd.rescue.sh backup```

Then reinstall it: (will install the recommended lnd  version - v0.10.4-beta)
```/home/admin/config.scripts/lnd.update.sh verified```

If you want to install the latest lnd version:(v0.11.0-beta):
```/home/admin/config.scripts/lnd.update.sh reckless```

Then restart Raspiblitz and check

jalal, 2020-09-08:
Hi @uniibu , I tried that, but no luck, same problem. I think that something is crashing systemd and it can no longer start the lnd service, or trying to start the lnd daemon is crashing systemd, I will have to check further...

rootzoll, 2020-09-09:
If you have your lnd rescue-backup (and of course your seed words & static channel backup) then you can try to clean the HDD (just keep the blockchain) and re-setup with a fresh 1.6 sd card. During setup LND choose "Old Wallet" and then import the rescue file.

To clean the HDD: ssh in and type `repair` the. choose `RESET-HDD`

jalal, 2020-09-10:
Well, it seems to have fixed itself after a second reboot. There was a message about doing an fsck which may have had an effect... I'll close this for now.

closed 2020-09-10 by jalal