raspiblitz/raspiblitz #1518: LNURL Fallback support in mobile wallets

rootzoll, 2020-09-08:
Example: https://twitter.com/notifications

- Make a list of LN mobile wallets to check LNURL Fallback is working.
- Ask not supporting wallets to add support.
- Make PR to https://github.com/fiatjaf/awesome-lnurl#wallets

frennkie, 2020-09-08:
Where should that Twitter notification link point to..? :-D

rootzoll, 2020-09-09:
corrected link

michaelWuensch, 2020-09-20:
Just for your intereat, I added support in Zap Android. Will be shipped with the next version.

rootzoll, 2020-10-17:
OK I tested with Zap-iOS and if I scan (under 'send') a plain LNURL QR code it is recognizing the LNURL ... but if I scan a LNURL fallback scheme I get a 'Unvalid Address Format' error.

rootzoll, 2020-10-17:
Asked for shockwallet here: https://github.com/shocknet/wallet/issues/300

rootzoll, 2020-10-17:
Tried to test lntxbot with an LNURL generated from lnbits.com ... when I was putting the LNURL into a fallback-scheme URL its was not working, while the plain LNURL string worked:
[image]

rootzoll, 2020-10-17:
Tried LNbits ... if I add a LNURL in the "Paste Request" form and it works. But it failes with a fallback URI - like this:

[image]

The response by LNbits is:

[image]

rootzoll, 2020-10-17:
On coinos.io entering a LNURL on "send" works, but a LNURL fallback scheme is rejected like this:
[image]

rootzoll, 2020-10-17:
Breez wallet was able to scan plain LNURL QR codes ... but failed on LNURL fallback-scheme with "QR code cannot be processed" warning (both on android & ios) - on this example: 
[image]

rootzoll, 2020-10-17:
Tested BluWallet with a plain LNURL QR code wored - BUT LNURL fallback scheme failed (closing scan without response) - for example:

[image]

rootzoll, 2020-10-17:
Zeus (just tested Android, but asume iOS is the same) its recognizing the LNURL fallback scheme - all good :)

rootzoll, 2020-10-18:
BLW worked both - LNUR plain & with fallback-scheme.

rootzoll, 2020-10-18:
Blixt Wallet was able to scan a plain LNURL but falied on LNURL fallback-scheme with 'QR sav errer: Code is not a valid Lightning invoice'.

fiatjaf, 2020-10-18:
https://github.com/hsjoberg/blixt-wallet/pull/246

rootzoll, 2020-10-18:
Opened also issue on Breez wallet: https://github.com/breez/breezmobile/issues/381

rootzoll, 2020-10-18:
ShockWallet is welcoming a PR and might take a look at it:
https://github.com/shocknet/wallet/issues/300

fiatjaf, 2020-10-18:
Another one here: https://github.com/BlueWallet/BlueWallet/pull/1993

fiatjaf, 2020-10-18:
ShockWallet here: https://github.com/shocknet/wallet/pull/302

fiatjaf, 2020-10-18:
It does work on https://t.me/lntxbot if you just paste the LNURL string directly without the command. I think I must improve documentation to make it explicit that you don't need to write commands when pasting invoices or lnurls.

fiatjaf, 2020-10-18:
Here is an attempt at Breez: https://github.com/breez/breez/pull/127

rootzoll, 2020-11-11:
So it really looks good now - I think we can close this issue already for v1.6.2 release:
[image]

Last on my list is Zap for iOS ... I think sending this reminder is all we can do.
https://twitter.com/rootzoll/status/1326542471875104768?s=20

closed 2020-11-11 by rootzoll