raspiblitz/raspiblitz #1529: Specter is using its own certificate instead of the one generated by raspiblitz

nolith, 2020-09-10:
I've tested the new specter with https://github.com/rootzoll/raspiblitz/pull/1502#issuecomment-684531724

Once ready the web server was using a custom selfsigned certificate for `www.fakeurl.com` instead of the one generated by raspiblitz with the local IP address.

/cc @openoms

rootzoll, 2020-09-10:
@frennkie didnt we moved this to the RaspiBlitz cert?

rootzoll, 2020-09-10:
adding to v1.6.1 for testing

rootzoll, 2020-09-10:
I checked ... its correct the lastest specter install is using the self created cert.

rootzoll, 2020-09-16:
> But if Spector itself creates it's own certificate then we an also deal with this. Nginx can present the configured cert (either from LND, from LetsEncrypt or user provided) and connect upstream to Spector (Validating against the local spector cert).

@frennkie yes Specter is creating/using its own cert and its working as it is right now - do you want to change anything on the nginx for v1.6.1 or should we push this to v1.7 to decide later

frennkie, 2020-09-16:
As we do not have a nginx config for specter in `1.6.0` I would not add one now for `1.6.1` because of this issue. Let's keep it as a "known issue" and address it in `1.7`.

k9ert, 2022-05-13:
For the reference: Removing the cert-generation on specter and preventing it to listen on the LAN is as simple as changing this line:
https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/config.scripts/bonus.specter.sh#L338

A proposal for a nginx configuration (pretty trivial) can be found here:
https://docs.specter.solutions/desktop/reverse-proxy/#basic-configuration