raspiblitz/raspiblitz #1532: cpufrequtils to reduce heat on Pi 4 CPU

ChuckNorrison, 2020-09-10:
i did a useful tweak to the cpu speed stepping via cpufrequtils. Thats the way how you can apply a speed stepping profile:

```
sudo apt-get install cpufrequtils
sudo cpufreq-set -g conservative
```

call `cpufreq-info` some times to get the current cpu clocks. It was at max frequency all the time in my case per default.

Maybe this is useful.

Following profiles are available:

```
ondemand (big steps automatic)
conservative (small steps automatic)
performance (highest clock possible allowed) 
powersave (lowest clock possible allowed)
``` 

ondemand was a bit too heavy in my observations. I did not miss any performance so far.

Marcoblue, 2020-09-27:
In my setup with a Pi 4 the temperature increases by 2°C with a conservative setting. Can someone recheck this please?

ChuckNorrison, 2020-09-27:
I think this is not worth to get into deeper. The default "ondemand" profile seems to work. After backup and running for some days, i currently got 30°C on my PI and default Profile. I isntalled the cpufreq-utils and set the profile to "conservative". After 1 Minute i meassured 32°C. After 10 Minutes it goes down to 30°C again.

conservative profile will more frequently try to set the best speed stepping. 
ondemand will not update the stepping in short timeframes, so i would prever the conservative profile here

Marcoblue, 2020-09-28:
Yes, I also believe that it is not really worth it. Here are my measurements with a Rapberry 4B (4GB) in a FLIRC case. 

- ondemand (standard): The CPU frequency is adapted to the current needs for computing power. The CPU frequency may be raised or lowered abruptly.

- conservative: The CPU frequency is adapted to the current requirements for computing power. However, the clock frequency is raised or lowered step-by-step. 

```
conservative:
400 MHz:42.15%, 467 MHz:3.92%, 560 MHz:3.39%, 700 MHz:3.60%, 933 MHz:6.38%, 1.40 GHz:40.57% (51°C)
```
```
ondemand:
400 MHz:40.54%, 467 MHz:9.02%, 560 MHz:2.53%, 700 MHz:2.59%, 933 MHz:15.04%, 1.40 GHz:30.28% (50°C)
```
```
powersave:
400 MHz:100%, 467 MHz:0%, 560 MHz:0%, 700 MHz:0%, 933 MHz:0%, 1.40 GHz:0% (48°C)
```
Also with only 400MHz it is not worth it.
(My Pi is limmited to 400-1400MHz instead of normal 600-1500MHz.)

ChuckNorrison, 2020-09-28:
Thanks for testing this. I would close this feature. 

If someone wants to check some cpu clocks, he will find this possibility here or at stackoverflow. I would not implement it as a feature for the blitz after tested without improvements.

closed 2020-09-28 by ChuckNorrison