raspiblitz/raspiblitz #1537: nginx service failing to run on boot

TrueFlow, 2020-09-10:
I was am having an issue accessing BTCPayServer via Tor &/or http on boot. I am running v1.6.0.

After running  - sudo systemctl status nginx

I get the error: 
Starting A high performance web server and a reverse proxy server...
nginx: [emerg] a duplicate listen [::1]:23002 in /etc/nginx/sites-enabled/btcpay_tor.conf:5
nginx: configuration file /etc/nginx/nginx.conf test failed
nginx.service: Control process exited, code=exited, status=1/FAILURE
nginx.service: Failed with result 'exit-code'.
Failed to start A high performance web server and a reverse proxy server.

After running -sudo service nginx restart my issue is resolved and I can connect to my BTCPayServer instance. I am unsure why the service is not running at the start.

rootzoll, 2020-09-14:
Will retest BTCPayServer install for v1.6.1 release.

rootzoll, 2020-09-14:
Also because of this merge: https://github.com/rootzoll/raspiblitz/pull/1539

rootzoll, 2020-09-16:
@TrueFlow I tested again for the v1.6.1 release and on my machine the BTCPay is running after a reboot from the beginning. Is this happening on your machine after every reboot or just after the first boot after the update?

rootzoll, 2020-09-16:
The merge #1539 looks fine .... marked for final testing.

frennkie, 2020-09-20:
@TrueFlow can you run this command and paste the output:

```
grep 2300 /etc/nginx/sites-available/*
```

syracusebitcoin, 2020-09-28:
I am running into this error too. my raspiblitz needed to reboot and after the reboot I can no longer log into BTCPay server, Thunderhub, or LNbits from my lan connection.

rootzoll, 2020-10-12:
OK tested BTCPay server for v1.6.1 release ... all testings look good. Closingt this issue.

closed 2020-10-12 by rootzoll