raspiblitz/raspiblitz #1538: Tor Zeus and LineageOS

woeisme, 2020-09-11:
I'm not certain this is a RaspiBlitz issue per se but is good to document somewhere as users might be able to find this. 

There seems to be a problem with using [Orbot VPN mode on LineageOS ](https://gitlab.com/LineageOS/issues/android/-/issues/2193). This will effect any user attempting to connect to Raspiblitz over Tor via ZAP or Zeus tutorials. 

There seems to be a fix listed [here](https://github.com/guardianproject/orbot/issues/151) thought it hasn't for me and requires root on the phone in question.

rootzoll, 2020-09-14:
If orbot app is not working - try IP2TOR as a fallback.

woeisme, 2020-10-01:
This is a non LineageOS update, on stock Android 10 I'm getting a Zeus error of:

> Error: stream was reset: INTERNAL_ERROR

woeisme, 2020-10-01:
ugh I'm an idiot!
I was using the wrong tor hostname!

closed 2020-10-01 by woeisme