raspiblitz/raspiblitz #1546: Optional LND update to 0.11.0

rootzoll, 2020-09-15:
The service release v1.6.1 wil be based on the same sd card build as v1.6 - just with updated scripts on top. So it will remain on install on LND 0.10.4 ... but we should offer the latest v0.11.0-beta as a optional-verified update in the "UPDATE" section.
https://github.com/lightningnetwork/lnd/releases/tag/v0.11.0-beta

rootzoll, 2020-09-15:
On running the verified update to LND 0.11.0 while checking the signing key .. it ran trhu but I got the following console output:
```
2020-09-15 10:14:32 (78.8 MB/s) - ‘/home/admin/download/pgp_keys.asc’ saved [6909/6909]

gpg: keybox '/home/admin/.gnupg/pubring.kbx' created
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: /home/admin/.gnupg/trustdb.gpg: trustdb created
gpg: key A96387A57CAAE94D: public key "Olaoluwa Osuntokun <laolu32@gmail.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
/dev/fd/63: line 13: pub: command not found
/dev/fd/63: line 14: 9769140D255C759B1EB77B46A96387A57CAAE94D: command not found
/dev/fd/63: line 15: syntax error near unexpected token `newline'
/dev/fd/63: line 15: `uid           Olaoluwa Osuntokun <laolu32@gmail.com>'
making sure the loop service is not running
*** INSTALL LIGHTNING LOOP ***
```

@openoms is this something we should look deeper into?

openoms, 2020-09-15:
seems like a gpg error, will test. 
Did not get this with this update script: https://github.com/openoms/lightning-node-management/tree/master/lnd.updates#update-lnd-to-v0110-beta   
Running on 0.11 on all of my nodes.

rootzoll, 2020-09-15:
This is the optional update script I am working on in the `1546lndupdate` branch: https://github.com/rootzoll/raspiblitz/blob/1546lndupdate/home.admin/config.scripts/lnd.update.sh

openoms, 2020-09-15:
Cannot reproduce on my 1.6 test node, seems all good:
https://pastebin.com/raw/tp6j6J7d

Do you have different gpg version?
```
admin@tBlitz:~ $ gpg --version
gpg (GnuPG) 2.2.12
libgcrypt 1.8.4
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Home: /home/admin/.gnupg
Supported algorithms:
Pubkey: RSA, ELG, DSA, ECDH, ECDSA, EDDSA
Cipher: IDEA, 3DES, CAST5, BLOWFISH, AES, AES192, AES256, TWOFISH,
        CAMELLIA128, CAMELLIA192, CAMELLIA256
Hash: SHA1, RIPEMD160, SHA256, SHA384, SHA512, SHA224
Compression: Uncompressed, ZIP, ZLIB, BZIP2
```

rootzoll, 2020-09-15:
Its the same gpg version ... will check if I can reproduce again.

rootzoll, 2020-09-15:
Was not able to reproduce this output again ... keep on eye on this on final testing.

rootzoll, 2020-09-15:
Implentation merged. First testing looks good. Test again on final testing.

ChuckNorrison, 2020-09-15:
I did this update and my system seems to be bricked... some infos while updating:

> --2020-09-15 22:34:59--  https://keybase.io/roasbeef/pgp_keys.asc
> Resolving keybase.io (keybase.io)... 34.234.80.85, 34.234.81.219
> Connecting to keybase.io (keybase.io)|34.234.80.85|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 6909 (6.7K) [text/plain]
> Saving to: ‘/home/admin/download/pgp_keys.asc’
> 
> /home/admin/download/pgp_keys 100%[=================================================>]   6.75K  --.-KB/s    in 0s
> 
> 2020-09-15 22:35:00 (72.6 MB/s) - ‘/home/admin/download/pgp_keys.asc’ saved [6909/6909]
> 
> gpg: WARNING: no command supplied.  Trying to guess what you mean ...
> gpg: key A96387A57CAAE94D: public key "Olaoluwa Osuntokun <laolu32@gmail.com>" imported
> gpg: Total number processed: 1
> gpg:               imported: 1
> /dev/fd/63: line 13: pub: command not found
> /dev/fd/63: line 14: 9769140D255C759B1EB77B46A96387A57CAAE94D: command not found
> /dev/fd/63: line 15: syntax error near unexpected token `newline'
> /dev/fd/63: line 15: `uid           Olaoluwa Osuntokun <laolu32@gmail.com>'
> 
> LCD turns white when shutdown complete.
> It will then reboot again automatically.
> -----------------------------------------------
> stop lnd - please wait ..
> stop bitcoind (1) - please wait ..
> Bitcoin Core stopping
> stop bitcoind (2) - please wait ..
> starting shutdown ...

this ends up in a root login on LCD screen. I can switch with `su admin` but permission and nothing seems to work anymore

rootzoll, 2020-10-07:
Test again because of merge of #1611

rootzoll, 2020-10-08:
OK looks good so far. Will be part of v1.6.1 - closing this issue.

rootzoll, 2020-10-11:
Opening due to #1647 - going for an enforced LND update with v1.6.1 and to recheck reckless update option.

rootzoll, 2020-10-15:
OK .. the new sd card for v1.6.1 comes now with LND 0.11.1. Closing this issue.

closed 2020-10-15 by rootzoll