raspiblitz/raspiblitz #1549: Update: Faraday 0.2.1

rootzoll, 2020-09-15:
https://github.com/lightninglabs/faraday/releases/tag/v0.2.1-alpha

For details in FAQ: https://lightning.engineering/posts/2020-09-15-faraday-accounting/

rootzoll, 2020-09-15:
Moving to v1.7 because the new feature needs LND 0.11.0 .. and that will become default in v1.7 - just optional upadte in v1.6.1

rootzoll, 2021-03-15:
OK v0.2.3 installs (see branch)

had on first try after install the problem that `/home/faraday/.faraday/${chain}net/faraday.macaroon` was nit created yet - appeared later on. To give the system the time to clean create tls/macaroon I activated force reboot is the user installs thru SERVICES menu.

Otherwiese looks good.

rootzoll, 2021-03-15:
merged inte dev. ok for final testing

openoms, 2021-03-15:
@rootzoll mind that the Lightning Terminal packages together Pool, Loop and Faraday with the CLI exposed as well. The separate install scripts can possibly be deprecated,

openoms, 2021-03-15:
@rootzoll mind that the Lightning Terminal packages together Pool, Loop and Faraday with the CLI exposed as well. The separate install scripts can possibly be deprecated.

rootzoll, 2021-03-16:
@openoms Should we drop the single installs already with the v1.7? Will RTL still work with LightningTerminal or does it need a single instance pool?

openoms, 2021-03-16:
I presume you mean Loop for RTL. The RPC interface is exposed with LiT as well juts differently. Need to test it, but it shoukd work.
the guys at LightningLabs are planning to favour the packed LiT and I think it is enough to keep LND separate.
I think we could take the individual installs out of the menu and recovery to avoid interference and confusion, but the scripts could stay.

closed 2021-04-13 by rootzoll