raspiblitz/raspiblitz #1551: "Optional LND Update to 0.11" results in a lockdown

ChuckNorrison, 2020-09-16:
After i updated my raspiblitz Software with the v1.6 Branch, everything worked fine and i did a successful reboot. When i executed the LND update via raspi menu, some errors where thrown:

```
--2020-09-15 22:34:59--  https://keybase.io/roasbeef/pgp_keys.asc
Resolving keybase.io (keybase.io)... 34.234.80.85, 34.234.81.219
Connecting to keybase.io (keybase.io)|34.234.80.85|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6909 (6.7K) [text/plain]
Saving to: ‘/home/admin/download/pgp_keys.asc’

/home/admin/download/pgp_keys 100%[=================================================>]   6.75K  --.-KB/s    in 0s

2020-09-15 22:35:00 (72.6 MB/s) - ‘/home/admin/download/pgp_keys.asc’ saved [6909/6909]

gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: key A96387A57CAAE94D: public key "Olaoluwa Osuntokun <laolu32@gmail.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
/dev/fd/63: line 13: pub: command not found
/dev/fd/63: line 14: 9769140D255C759B1EB77B46A96387A57CAAE94D: command not found
/dev/fd/63: line 15: syntax error near unexpected token `newline'
/dev/fd/63: line 15: `uid           Olaoluwa Osuntokun <laolu32@gmail.com>'

LCD turns white when shutdown complete.
It will then reboot again automatically.
-----------------------------------------------
stop lnd - please wait ..
stop bitcoind (1) - please wait ..
Bitcoin Core stopping
stop bitcoind (2) - please wait ..
starting shutdown ...
```

The reboot results in a root login without network connectivity. I was able to solve it temporary with daemon-reload and restart dhcpcd service here. After reboot i get back to that point.

Any suggestions whats next?

ChuckNorrison, 2020-09-16:
I found a fresh lnd-rescue file and currently i will take an image of the rootfs partition for analyze. I will try to rebuild the sdcard and restore my data from hdd and the lnd-rescue file as proposed in the docs. Hopefully this will work.

rootzoll, 2020-09-21:
Good to hear that you were able to recover.
> // i observed a permission denied error while patching to current version via patch menu:

This one will be gone with v1.6.1 update

> Seems the root issue of booting is the delayed network interface behaviour which needs to be resolved. Further the admin user lost his permissions while trying some ssh or scp connections. How can this happen?

I am not sure ... sounds like a very strange and edgy bug. Last time we got such reportings it was some hardware related stuff like wit the shim fan. I am closing the issue (while #1546 stay open for testing).

closed 2020-09-21 by rootzoll