raspiblitz/raspiblitz #1554: Export the Connection String for BTCPayserver

Animalmother9000, 2020-09-16:
Would it be possible to implement an easy to use function for exporting the "Connection String" needed for any BTCPayserver to use the RaspiBlitz as underlying Lightning Node?

For example I use a "Third Party Hosted" BTCPayserver and want to accept Lightning payments.

This is the needed Connection String:
[image]

rootzoll, 2020-09-17:
Does the macaroon needs to be admin? Will just invoice-macaroon work? If you just want to "receive" payments having a "not admin" macaroon would be the best security option.

rootzoll, 2020-09-17:
@Animalmother9000 this should now work - give it a try.
Run on a v1.6.0 RaspiBlitz MENU > UPDATE > PATCH > PATCHMENU > PATCH and reboot
Then go: MENU > LNDCRED > EXPORT > BTCPAY Server

This should give you the connection string.for Lightning Payments. For the xPub on on-chain payments with BTC_pAy server its still recommended to use the xpub of a hardware wallet here (more security).

Also its advised to loop-out funds regularly from the RaspiBlitz ... because the BTCPay server has still the admin macaroon and if someone hacks that server then could steal the funds from RaspiBlitz. 

I tried using the "invoices.macaroon" but that did not worked with BTCPay. If you can find out more (what macaroon or what macaroon to bake) that can be used to receive but not to send with BTCPay ... that would be a great impovement.

Animalmother9000, 2020-09-19:
I subscribed for a Ip2Tor bridge which worked smoothly. After that I could export the string and tried to connect the node with the BTCPayserver. I tried it several times and mostly I got an error. I attached the message below. 

[image]

But sometime it worked and the node connected to the BTCPS. 

After that I made some test orders and the BTCPS correctly issued a Lightning invoice. I tried to pay it with three different wallets but was not able to pay it. I have to figure out the reason for that and will do more tests with different nodes and wallets. 

Now I also tried it with another BTCPS but was not able again to connect. I always get the timeout error. Maybe it is because of delays in the Tor network.

rootzoll, 2020-09-20:
@frennkie can you check why this is working sometimes and sometimes not? 

@Animalmother9000 can you check your Tor logs (see nyx) and check if there is something similar to this: https://github.com/rootzoll/raspiblitz/issues/1449#issuecomment-679107531

frennkie, 2020-09-20:
> can you check why this is working sometimes and sometimes not?

I included his port in my IP2Tor endpoint monitoring. I guess it's the same issues with Tor rendezvous points.

rootzoll, 2020-10-12:
The basic export will be part of v1.6.1 release ... but will move this ticket to the v1.7 milestone to improve with better cut macaroon and automatic external reachability.

rootzoll, 2020-10-12:
For baking a macaroon see: https://github.com/btcpayserver/btcpayserver/issues/1845

rootzoll, 2020-12-17:
TODO: Add Bakemacaroon for v1.6.3 release

rootzoll, 2020-12-24:
OK - changed to macaroon baking. @Animalmother9000 if you take a v1.6.2 and use UPDATE > PATCH > PATCH-MENU > PATCH then you should have the new baked macaroon. If you can test that would be great.

rootzoll, 2021-01-12:
This will be part of v1.6.3 - closing for v1.6.3 release.

closed 2021-01-12 by rootzoll