raspiblitz/raspiblitz #1555: Issues syncing blockchain

swidgen, 2020-09-16:
after wiping hard drive and flashing the SD card, upon first reboot raspiblitz is not automatically syncing the blockchain.  Screen says `Error found in logs` so instead I try repair and resync a new blockchain.  This seems to work but consistently after about 8% the sync fails and stops.  this is the debug Logs.  I've tried reflashing, wiping and syncing a few times and this problem persists.  I've also looked through about 10 pages of issues on github but can't find the exact same problem, anyway after several pages it's much older versions
`* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.6.0
chainnetwork: bitcoin / main
 22:08:16 up  1:48,  2 users,  load average: 1.29, 2.46, 2.87

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2020-09-16 22:08:15 BST; 747ms ago
  Process: 26683 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 26685 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=1/FAILURE)

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2019-02-14 10:11:59 GMT, end at Wed 2020-09-16 22:08:16 BST. --
Sep 16 22:08:15 raspberrypi systemd[1]: bitcoind.service: Service RestartSec=30s expired, scheduling restart.
Sep 16 22:08:15 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 2.
Sep 16 22:08:15 raspberrypi systemd[1]: Stopped Bitcoin daemon.
Sep 16 22:08:15 raspberrypi systemd[1]: Starting Bitcoin daemon...
Sep 16 22:08:15 raspberrypi bitcoind[26685]: Error: Cannot write to data directory '/home/bitcoin/.bitcoin'; check permissions.
Sep 16 22:08:15 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Sep 16 22:08:15 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Sep 16 22:08:15 raspberrypi systemd[1]: Failed to start Bitcoin daemon.
1600290473 ERROR: Error found in Logs
1600290477 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-09-16T21:06:03Z Pre-allocating up to position 0xa00000 in rev00159.dat
2020-09-16T21:06:03Z UpdateTip: new best=00000000000000001cdb384d7241827782aa5cf33371567e35c3e08226ab95c2 height=312208 version=0x00000002 log2_work=79.865834 tx=43125604 date='2014-07-24T02:59:06Z' progress=0.076301 cache=606.3MiB(5098447txo)
2020-09-16T21:06:03Z UpdateTip: new best=0000000000000000201ca74002bb4db07e1db575e9cdc9cfb1e5cd45be108d36 height=312209 version=0x00000002 log2_work=79.865931 tx=43125732 date='2014-07-24T03:05:10Z' progress=0.076301 cache=606.3MiB(5098558txo)
2020-09-16T21:06:03Z UpdateTip: new best=0000000000000000348573c4dbeef0365aef2c9f58b60c8e4cb45510cf72d927 height=312210 version=0x00000002 log2_work=79.866029 tx=43126478 date='2014-07-24T03:21:30Z' progress=0.076302 cache=606.4MiB(5099200txo)
2020-09-16T21:06:03Z UpdateTip: new best=000000000000000026d87d5e405b0f7d7ac9b1ce4327e1cbb0bc540674d62937 height=312211 version=0x00000002 log2_work=79.866126 tx=43126606 date='2014-07-24T03:36:42Z' progress=0.076303 cache=606.4MiB(5099273txo)
2020-09-16T21:06:03Z UpdateTip: new best=000000000000000030cc133929a1d3dce5fe2c404199e7239c74ef302f4d7d4e height=312212 version=0x00000002 log2_work=79.866224 tx=43126978 date='2014-07-24T03:31:12Z' progress=0.076303 cache=606.5MiB(5099653txo)
2020-09-16T21:06:03Z UpdateTip: new best=0000000000000000340f8ccbb924b041f8f876e70fb1f651c5b74053ba58efd4 height=312213 version=0x00000002 log2_work=79.866321 tx=43127717 date='2014-07-24T03:50:43Z' progress=0.076305 cache=606.5MiB(5099929txo)
2020-09-16T21:06:03Z UpdateTip: new best=0000000000000000293d8c9321f16bd6fa83045688064286762ac313f9db4191 height=312214 version=0x00000002 log2_work=79.866419 tx=43128484 date='2014-07-24T04:10:39Z' progress=0.076306 cache=606.6MiB(5101031txo)
2020-09-16T21:06:03Z UpdateTip: new best=00000000000000002dd7f7bfd9d830cdfa375603b914f3c28be9df94c7bdd866 height=312215 version=0x00000002 log2_work=79.866516 tx=43128722 date='2014-07-24T04:16:32Z' progress=0.076306 cache=606.6MiB(5101075txo)
2020-09-16T21:06:03Z UpdateTip: new best=0000000000000000133fe0b4636a1adfcabdc58f6a687f87b694b508cd166f88 height=312216 version=0x00000002 log2_work=79.866614 tx=43129010 date='2014-07-24T04:25:37Z' progress=0.076307 cache=606.6MiB(5101043txo)
2020-09-16T21:06:03Z UpdateTip: new best=000000000000000009be6cb34cf84cbe0de868521d86281c22279b9213b47b3a height=312217 version=0x00000002 log2_work=79.866711 tx=43129049 date='2014-07-24T04:27:03Z' progress=0.076307 cache=606.6MiB(5101148txo)
2020-09-16T21:06:03Z UpdateTip: new best=00000000000000003263330cc748de3253da7f90dfc61b7f23f25c81a5314195 height=312218 version=0x00000002 log2_work=79.866809 tx=43129473 date='2014-07-24T04:34:42Z' progress=0.076308 cache=606.6MiB(5101287txo)
2020-09-16T21:06:04Z UpdateTip: new best=00000000000000000608e7d5c7bfb01df67bc9047480cc452d0f60b01f6031d9 height=312219 version=0x00000002 log2_work=79.866906 tx=43129833 date='2014-07-24T04:42:27Z' progress=0.076308 cache=606.6MiB(5101432txo)
2020-09-16T21:06:04Z UpdateTip: new best=000000000000000010b42150a6ba7d29cf66bb3f8eadaabb98e3959decc077a8 height=312220 version=0x00000002 log2_work=79.867003 tx=43130055 date='2014-07-24T05:08:31Z' progress=0.076309 cache=606.7MiB(5101528txo)
2020-09-16T21:06:04Z UpdateTip: new best=0000000000000000397e178329db43da460cc5cd7f49332937879f5a4298a389 height=312221 version=0x00000002 log2_work=79.867101 tx=43130403 date='2014-07-24T04:57:14Z' progress=0.076309 cache=606.7MiB(5101573txo)
2020-09-16T21:06:04Z UpdateTip: new best=0000000000000000364591d158b2ad540bc36be8fcca23aed41194b44bc32d27 height=312222 version=0x00000002 log2_work=79.867198 tx=43130659 date='2014-07-24T05:16:28Z' progress=0.076310 cache=606.7MiB(5101696txo)
2020-09-16T21:06:04Z Pre-allocating up to position 0x6000000 in blk00159.dat
2020-09-16T21:07:14Z LevelDB read failure: IO error: /mnt/hdd/bitcoin/chainstate/001161.ldb: Input/output error
2020-09-16T21:07:14Z Fatal LevelDB error: IO error: /mnt/hdd/bitcoin/chainstate/001161.ldb: Input/output error
2020-09-16T21:07:14Z You can use -debug=leveldb to get more complete diagnostic messages

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: resources) since Wed 2020-09-16 22:08:15 BST; 1s ago

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Thu 2019-02-14 10:11:59 GMT, end at Wed 2020-09-16 22:08:16 BST. --
Sep 16 20:25:16 raspberrypi systemd[1]: Starting LND Lightning Daemon...
Sep 16 20:25:16 raspberrypi systemd[1]: Started LND Lightning Daemon.
Sep 16 22:07:14 raspberrypi lnd[3193]: Unable to determine if wallet is synced: Post http://127.0.0.1:8332: EOF
Sep 16 22:07:14 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Sep 16 22:07:14 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
Sep 16 22:08:14 raspberrypi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Sep 16 22:08:14 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 1.
Sep 16 22:08:14 raspberrypi systemd[1]: Stopped LND Lightning Daemon.
Sep 16 22:08:15 raspberrypi systemd[1]: lnd.service: Failed to load environment files: No such file or directory
Sep 16 22:08:15 raspberrypi systemd[1]: lnd.service: Failed to run 'start-pre' task: No such file or directory
Sep 16 22:08:15 raspberrypi systemd[1]: lnd.service: Failed with result 'resources'.
[... 69 lines cut ...]
*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
raspberryPi=4
powerFAIL=0
powerWARN=0
powerMIN='860000 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='5400 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.0.19'
tempCelsius='55.0'
uptime=6524
upsStatus='OFF'
startcountBlockchain=3
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=1
lndActive=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0`

ChuckNorrison, 2020-09-17:
your root cause seems to be permissions and/or missing files:
`Sep 16 22:08:15 raspberrypi bitcoind[26685]: Error: Cannot write to data directory '/home/bitcoin/.bitcoin'; check permissions.`
and 
`Sep 16 22:08:15 raspberrypi systemd[1]: lnd.service: Failed to load environment files: No such file or directory`

i would try to check if files are missing or permissions are broken. If permissions are broken, i did well with the lnd-rescue*.tar.gz and rebuild the sd card and fresh install. Maybe someone else can solve this much easier with a chown -r

rootzoll, 2020-09-21:
@swidgen >This seems to work but consistently after about 8% the sync fails and stops. this is the debug Logs.

Are you using the recommended SSD case from the RaspiBlitz shopping list? We had reports in the past that some cases dont support all needed standards and the connection to the drive freezes.

swidgen, 2020-09-21:
ya, that was the problem, just had to switch from USB3 port to USB2 port and it's running fine now.

closed 2020-09-21 by swidgen